jopohl/urh #222: urh crashes when capturing data from RTL-SDR on macOS Sierra

cyoung, 2017-04-03:
macOS Version: 10.12.4
Python version: 3.4.5
urh version: 1.6.2.0 / https://github.com/jopohl/urh/commit/f5d964d3a6baddfd14f63b7d603fb17c126f4b4e

Cloned the source, ran `sudo python3.4 setup.py install`. Couldn't figure out where "urh" was supposed to be installed in my path, so I ended up running it from the git clone'd directory with the following results.

There is an RTL-SDR dongle plugged in. `rtl_test`, `gqrx`, etc work fine with it.

The crash happens when hitting the "record" button on Spectrum Analyzer. It also seems to happen anywhere else the program attempts to access the RTL-SDR to capture data.

Output:

     $ ./main.py
     Will not regenerate UI, because script cant be found. This is okay in release.
     Using modules from /private/tmp/urh/src
     [INFO] Initialized receiving buffer with size 0.76MB
     [INFO] RTLSDR: Starting RX Mode
     Traceback (most recent call last):
       File "/private/tmp/urh/src/urh/controller/SpectrumDialogController.py", line 82, in on_start_clicked
         self.device.start()
       File "/private/tmp/urh/src/urh/dev/VirtualDevice.py", line 417, in start
         self.__dev.start_rx_mode()
       File "/private/tmp/urh/src/urh/dev/native/Device.py", line 289, in start_rx_mode
         self.receive_process.start()
       File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/process.py", line 105, in start
         self._popen = self._Popen(self)
       File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/context.py", line 212, in _Popen
         return _default_context.get_context().Process._Popen(process_obj)
       File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/context.py", line 274, in _Popen
         return Popen(process_obj)
       File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/popen_spawn_posix.py", line 33, in __init__
         super().__init__(process_obj)
       File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/popen_fork.py", line 21, in __init__
         self._launch(process_obj)
       File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/popen_spawn_posix.py", line 48, in _launch
         reduction.dump(process_obj, fp)
       File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/reduction.py", line 59, in dump
         ForkingPickler(file, protocol).dump(obj)
     _pickle.PicklingError: Can't pickle <function RTLSDR.receive_sync at 0x10cddd8c8>: attribute lookup receive_sync on urh.dev.native.RTLSDR failed
     /opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/multiprocessing/semaphore_tracker.py:129: UserWarning: semaphore_tracker: There appear to be 2 leaked semaphores to clean up at shutdown
       len(cache))
     Abort trap: 6

jopohl, 2017-04-03:
I just checked on OSX 10.12 and I was not able to reproduce this. Does the problem persist after restarting the application? The error ``` attribute lookup receive_sync on urh.dev.native.RTLSDR failed ``` indicates, that it can't find the ``` receive_sync ``` function which is written in plain python, so it should be found in any case.

cyoung, 2017-04-03:
It does persist after restarting the application. I first encountered the issue with the pip installed version, then tried this method and it also had the same issue.

jopohl, 2017-04-03:
I see that you are using a quite old python version (3.4). Is there any specific reason for this? If not, you could try updating to [Python 3.6](https://www.python.org/downloads/mac-osx/) which runs on my OSX 10.12 and works fine with RTL-SDR.

cyoung, 2017-04-03:
It's what was installed on the system. The README has minimum requirement "Python 3.4+".
I've just installed Python 3.6.1 and everything is working fine now, so that did it. Thank you.

jopohl, 2017-04-03:
You're welcome @cyoung . Glad it works now! I added a hint for this to the readme.

closed 2017-04-03 by jopohl