jopohl/urh #223: Fix for low resolution / small screen size

ngmacha, 2017-04-03:
(This is copied from issue #220 for follow up)

The initial issue: 
The LCD display on my Pine64 board is having only resolution of 1024x600 while URH and most of its dialogs opens by default with the height large than 600 pixels. There are no vertical scroll bars, making the buttons and options at the bottom inaccessible - so every time opening a new dialog I have to resize them. Is it possible to enable vertical scrolls, or set the default screen size to which the windows would fit to?

@jopohl : I just wrote a fix 6c02676 for low resolutions. URH should be way better accessible on your screen now! You can check out master to have access to this fix or just wait until I make the next release.

jopohl, 2017-04-03:
Did the fix work for you?

ngmacha, 2017-04-03:
I've tried to run from the master, and it looks much much better. However the Armbian system menu uses a band at the top of screen, the dialogs still have to be resized to reduce the heights to show full size (for example the horizontal scrolls under the Spectrum Analyzer). 

Where the default/max screen size is set, is it hard-coded? Maybe you add the option like max_width, max_height under urh.ini?

jopohl, 2017-04-03:
I just fixed this by saving the geometry for the dialogs, so you only have to resize them once. Could you check out the branch ``` restore_geometry ``` and tell me, what you think? We can merge it into master, when it is fine.

ngmacha, 2017-04-03:
Just tested - the sizes of the dialog retains within the URH session, but if I close the URH main window and then relaunch URH, they revert to default big ones.

However... I ran `git clone https://github.com/jopohl/urh --branch restore_geometry` and then ran  main.py from under the src/urh subfolder - is this a correct way to test? Or I have to build it fully?

jopohl, 2017-04-03:
It should survive an application restart. The window settings are stored under ~/.config/urh.ini. Is another URH process running, that may overwrite changes to this file?

ngmacha, 2017-04-03:
Yes, you are right, the urh.ini was owned by root and user doesn't have write rights - I deleted the file and rerun URH and now the dialogs sizes are retained!! Thank you for very quick response, I think this could be merged to master.... 

And BTW my issue #224 of RTL-SDR and HackRF reverting to disabled is fixed too - looks like due to non-writable urh.ini!!

jopohl, 2017-04-03:
Nice! I merged it into master, so I will close here.

closed 2017-04-03 by jopohl