jopohl/urh #225: How to decode this kind of modulation?

ngmacha, 2017-04-03:
Hi
I have a device (433MHz wireless temperature sensor from LaCrosse) that transmit series of 44 long and short pulses, as can be seen in this screen shot:
[image]

Zooming in to some of the pulses:
[image]
 
What kind of modulation is this, and how to decode it in URH? Just guessing that long pulses are ones and short are zeroes, all three modulation types in URH (ASK, FSK and PSK) are not giving correct bit strings...

Thank you for advise.

jopohl, 2017-04-03:
You can try setting the bit length to the length of your zero pulse and modulation type to ASK. 

After that, go to analysis, choose decoding combobox -> "..." entry. (this opens a new dialog) In this dialog create a new decoding dragging "Substiution" from base functions to your new decoding.

ngmacha, 2017-04-14:
Sorry Johannes, didn't have time to try this yet...

Mac Ha - Sent from iPhone

Closed #225<https://github.com/jopohl/urh/issues/225>.

jopohl, 2017-04-14:
Oh sorry I thought this was resolved.

ngmacha, 2017-04-20:
Thank you for reopening this. Only now I got some time to try it out. As per your advise, I set the modulation to ASK and bit length to 70, error tolerance 10, and get this:
[image]
Then defining the following Decoding substitution:
[image]
I finally get this, which correctly represents the sequence of long and short pulses:
[image]

However now is how to make URH to show the messages, which are the sequence of 43 bits - with the pauses of >30000 samples in between? Which Decoding mechanism to use to join the bits  that have pauses in between them less than say 5000 samples into one message? Will have to try further...

andynoack, 2017-04-20:
Can you upload your signal? I think, there is some experimentation with different settings necessary.

ngmacha, 2017-04-20:
Here is the sample
[RTL-SDR 433_920MHz 1MSps 1MHz_complex.zip](https://github.com/jopohl/urh/files/945012/RTL-SDR.433_920MHz.1MSps.1MHz_complex.zip)

Looks very simple, so maybe you'll solve it right away - excuse me a novice in SDR, but I thought there should be some kind of "pulse length modulation/encoding"?...

andynoack, 2017-04-20:
Hey @ngmacha 
I just changed the function of the Substitution encoding a little bit. Get the latest URH and then try:
* Bit length = 1100
* Modulation = ASK
* Encoding: Substitution (1 row): "10"->"1"
Does that work for you?

ngmacha, 2017-04-21:
No, that unfortunately didn't work - I've tried this before. Somehow with this substitution the standalone 0 and 1 are thrown away, even looking at the test here:
[image]

So the result I got are just 23 ones... Looks like the Substitude function triggers error if encounters sequences that doesn't match any "From" patterns, and throw them away. Should it be the other way instead, to pass them unchanged?...

URH is 1.6.23...

andynoack, 2017-04-21:
I did not make a release for the fix. Did you use the current git version of URH?

ngmacha, 2017-04-21:
Oh, no, which git version should I use?..

andynoack, 2017-04-21:
There will be a release v1.6.2.4 in some minutes, where the fix will be included :)

ngmacha, 2017-04-21:
Great, just updated and it works!! So it's time for more capturing and hacking!!

Thank you very much, a great software and great support!

closed 2017-04-21 by ngmacha

andynoack, 2017-04-21:
For the next time: I implemented a Morse Code encoding that might help with those signals. When using the new encoding, you probably don't need to experiment with bit length anymore.

ngmacha, 2017-04-22:
That will be great!! When can we expect that?

Mac Ha - Sent from iPhone

andynoack, 2017-04-22:
It is already there :) However, not in a release, so you would need to use the current git version or wait for v1.6.2.5 (which is not terminated yet).

scirusvulgaris, 2020-04-17:
A bit late but you never know might help someone, in order to decode correctly a LaCrosse weather signals. First, do autodetect, then fine tune the Sample/Symbol, for me it was a bit longer then it should be. Second go to analyze, here i would advise substituting 1100 > 1 and 1000 to 0. It was my case. Remember, The key is to check how many bits you have till next signal when you demodulate with ASK.
Cheers

commit b74707f8 (2017-04-21): Encoding: Add Morse Code encoding (fix #225)