jopohl/urh #232: no Frequency Correction option for SDR

roema, 2017-04-12:
Hi!

I have nothing found about frequency correction for SDR (hackrf).
Is it possible to add a field in the options for SDR to add the correction value in ppm?
 Or is there another option to correct that?

Thanks for you answer!

Manu

jopohl, 2017-04-12:
Hi @roema ,

unfortunately, there seems to be no method for setting frequency correction in hackrf driver: https://github.com/mossmann/hackrf/blob/master/host/libhackrf/src/hackrf.h

roema, 2017-04-14:
Hi @jopohl!

Thanks for your message! It's no problem!
UHR is a very grateful tool! Today i have successfully cloned a garge door transmitter (ASK) an the remote car key from my suzuki (FSK) in few minutes!

jopohl, 2017-04-14:
Glad to hear that @roema - it is a pleasure to see that URH ist helpful.

closed 2017-04-14 by jopohl