jopohl/urh #233: [LimeSDR Beta] No way to capture data

romeojulietthotel, 2017-04-13:
In the terminal window where I launch Urh there is this error:

```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

```
That error is displayed after Urh reports that it has connected to the LimeSDR. But there's no data displayed. I tried all three capture/record/spectrum and none worked.
The LimeSDR has the latest firmware/gateware and works fine with all my other SDR apps.

Also I had to copy the lime header by hand into the Urh install to build the native driver.

FWIW I just did this:

`cp -rpv /limesuite/include/lime lib/python3.6/site-packages/urh-1.6.2.0-py3.6-linux-x86_64.egg/urh/dev/native/includes/`

jopohl, 2017-04-13:
The fact, that you had to copy the header indicates that you somehow used an old version, because the header files are included. Could you try running:
```
git clone https://github.com/jopohl/urh
cd urh
git checkout limesdr
./src/urh/main.py
```

romeojulietthotel, 2017-04-13:
I actually removed my old git clone and did a new one before filing this report.
I will try `./src/urh/main.py`
I ran this install command, is this wrong:
`python setup.py install --without-hackrf --without-rtlsdr`

(BTW even with this install command when building for native I saw errors about no hackrf or rtl-sdr files)

romeojulietthotel, 2017-04-13:
I tried the `./src/urh/main.py` and was able to add the native LimeSDR support. Restarted as intructed but there was again no LimeSDR device support.

jopohl, 2017-04-13:
Can you be more specific about this? Furthermore, you can run:
```
python tests/TestLimeSDR.py
```
to run some automated tests for your LimeSDR so we can see what is missing. Make sure, you pull first, because I just updated the test.

romeojulietthotel, 2017-04-13:
Pulled and ran and have this:
(I didn't paste the hexbytes that preceded the below output)

```
TuneVCO(SXR) - VCO too low
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted
```

I run the latest limesuite git master. 

LimeUtil --info and --timing and --find run with no errors.

jopohl, 2017-04-13:
So you do not even see the device list? Is your LimeSDR connected in another application?

jopohl, 2017-04-13:
If you see a long hex byte list, that means it could receive at least the first time. I suspect it could run out of memory. Could you set the the fifo size in urh/dev/native/LimeSDR.py:12 a bit lower, e.g. to 262144?

jopohl, 2017-04-13:
I just fixed the FIFO sizes in 5e6230baa4b14f92ae84d0bf9b1ae4dece684243. Works fine with my LimeSDR, can you test this new commit?

romeojulietthotel, 2017-04-13:
Pohl you're too fast for me!!

I set to 262144 and I am now rx'ing data.

Also here is where I found bad_alloc but have not looked at the code:

grep -n bad_alloc src/ConnectionSTREAM/ConnectionSTREAMing.cpp
327:    catch (const std::bad_alloc &ex)
554:    catch (const std::bad_alloc& ex) //not enough memory for buffers

jopohl, 2017-04-13:
Thanks for your investigation, this is a proof for my suspicion and explains why the lower FIFO size works. :+1: 

Could you also test the commit I made? I set the FIFO size a bit higher (to 1048576 instead 262144) to avoid buffer overruns, but I expect this to work as well.

romeojulietthotel, 2017-04-13:
Yes, that commit is working fine. I will file a separate report for a different issue.

romeojulietthotel, 2017-04-13:
So the huge FIFO size prevented data from displaying since the buffer had to fill first then flush to the display?  It seemed like the 262144 setting allowed the data to seen sooner while the 1048576 was just a little longer wait before data was displayed. Might be a nice tunable feature or maybe I am just wrong and misperceived the delay. Not sure how the FIFO is hooked up.

jopohl, 2017-04-13:
I am not entirely sure how the FIFO is implemented by the LimeSDR driver but I think it needs to be initialized in some way. That's why a larger FIFO takes up some more time. The downside of a smaller FIFO is that it may loose samples for high sample rates.

romeojulietthotel, 2017-04-13:
LimeSDR owners will be happy now this is fixed. It does work and maybe the way to handle the buffer is to display a busy wheel/beachball/progress bar, etc. so they know something's happening. But it's easy for me to make suggestions and I have little idea how difficult that would be to add.

jopohl, 2017-04-14:
I just added some information for this to the device log in 368b0a082031a54daf1199095c5f3ef7bd78fd77. I will close this one.

closed 2017-04-14 by jopohl