jopohl/urh #236: [LimeSDR] Native build leaving temporary files.

romeojulietthotel, 2017-04-14:
Lots of files left behind after the native build runs. I have found many in /tmp/*.c, /tmp/native_extension and found .o files in limesuite/include/tmp

Some may be due to initial problems getting the native extension to build but the ones in /tmp appear always it seems.

jopohl, 2017-04-15:
Could you explain why this is an issue for you? In my case, the ``` /tmp ``` directory gets cleared after reboot.

romeojulietthotel, 2017-04-16:
Potentially not a problem for only me. 

Here's one scenario where this would break things:

Rita has her own python install (virtualenv or full install), she installs URH and the temp files for the native driver are written to /tmp

Tina has her own python install (virtualenv or full install), she installs URH and the temp files for the native driver are written to /tmp but the process fails because Rita's files are in /tmp

Maybe it's okay to leave them but they should all then be written into a subdir of /tmp/   e.g. /tmp/rita-urh

There are other reasons to remove them altogether after building:

- Some systems do not clear /tmp on reboot
- Some systems rarely reboot and /tmp begins to be cluttered over time
- Some systems if put /tmp into ramfs or similar volatile filesystem that has space limits/constraints

jopohl, 2017-04-16:
Thanks for the explanation. I will take care of this.

closed 2017-04-17 by jopohl

romeojulietthotel, 2017-04-17:
Still one file remains called /tmp/native_extensions.

jopohl, 2017-04-17:
When does this file gets created? I rebuild the device extensions via options and do not have such a file. What is its content?

romeojulietthotel, 2017-04-17:
Have not looked into the file yet. I see it written here:

https://github.com/jopohl/urh/blob/master/src/urh/controller/OptionsController.py#L385

jopohl, 2017-04-17:
Thanks, I fixed that.

commit cda150ba (2017-04-17): fix #236