jopohl/urh #238: Shortcut to x axis zoom

vsboost, 2017-04-15:
Hi there,

I normally use a surface pro tablet and was wondering if you could possibly add a shortcut button to do "x axis" zoom or even a slider.  Unless i plug a mouse into the tablet i find it difficult to zoom in to the signal.

Thanks again.

vsboost, 2017-04-16:
Using the mouse scroll wheel seems to keep the signal centered which is good.

jopohl, 2017-04-16:
Hey @vsboost ,

I just fixed the keyboard zoom, it now behaves like the mousewheel zoom and scrolls to the position of the mouse cursor. What do you think about that?

vsboost, 2017-04-16:
Fantastic, thanks for this.

vsboost, 2017-04-17:
Sorry ill just re open this one because i noticed that when i select a signal and crop to new selection i can no longer use the Ctrl + to zoom in

jopohl, 2017-04-17:
I could not reproduce this one. Can you give more details? What I did was: Open a signal, select some samples, right click -> "crop to selection", press Ctrl ++ and Ctrl +-, which worked for me.
How many samples were left in your signal after cropping? The zoom is disabled for signals with less than 300 samples.

vsboost, 2017-04-17:
Its a strange one, what i found that if i use the touchpad on my tablet to make a selection and right click and crop to selection the new signal that is displayed wont let me zoom using Ctrl +, however if i do the exact same with a mouse it will let me zoom.
Workaround i just discovered is i need to select the new signal and make a selection of any width then i can use the Ctrl + to zoom.

Hope that makes sense, it doesn't bother me now that i know how to make it work, so if you cant reproduce just close this one.

Thanks again

jopohl, 2017-04-17:
Is it possible, that the new signal just loses the focus? In this case, it should be enough to make a single left click inside the new signal to make zooming work again.

vsboost, 2017-04-17:
This would be the case.

Cheers

closed 2017-04-17 by vsboost