jopohl/urh #240: 'Dark' Theme for Windows

EcstaticMark, 2017-04-16:
Hello, could an option for a 'dark' theme be added to the Windows version?

I'm spending a  lot of time staring at the URH screen and I'm going blind. :)

jopohl, 2017-04-16:
Hi @EcstaticMark ,

I am with you, I also prefer working with a dark theme. :+1: 

I just added an option for this under Options -> View. Please pull latest master and tell me what you think.

EcstaticMark, 2017-04-16:
I ran 
C:\WINDOWS\system32>pip3 uninstall urh
followed by
C:\WINDOWS\system32>pip3 install urh
I also tried 
C:\WINDOWS\system32>pip3 install urh --no-cache-dir
C:\WINDOWS\system32>pip3 install --upgrade urh
and 
C:\WINDOWS\system32>pip3 install --upgrade urh --no-cache-dir 
after each uninstall.

I still can't find an option for the dark theme. Any idea what I'm doing wrong?

jopohl, 2017-04-16:
Its currently not released, you can try it out with:
```
git clone https://github.com/jopohl/urh/
cd urh/src/urh
./main.py
```

jopohl, 2017-04-16:
I just released version 1.6.2.3, so you can also try again with pip3.

EcstaticMark, 2017-04-16:
Ahh, much better. I won't need these welding goggles any more. 👍 

Thanks!

jopohl, 2017-04-16:
Hehe no problem, you're welcome!

closed 2017-04-16 by jopohl