jopohl/urh #241: Ability to add HEX / ASCII values to fuzzed messages

vsboost, 2017-04-17:
Hi there,

Was wondering if its possible to add Range / Boundaries / Random data in hex / ascii ?

Cheers

andynoack, 2017-04-17:
I do not know whether I understand you right, but I think you can already do that by changing the view type in the generator before (bottom right)

vsboost, 2017-04-17:
What i mean is the fuzzed values are only decimal.

andynoack, 2017-04-17:
Ok, so you would like to see the border/... values in the current view type (hex, ascii) instead of values converted to decimal?

vsboost, 2017-04-17:
Correct

andynoack, 2017-04-17:
I think that will be a larger amount of work as we would need special input boxes for that. I am not sure if this really improves the useability (especially for ascii!) or makes it even worse (as you do not see the number of fuzzed values on the first sight). We will discuss that!

vsboost, 2017-04-17:
Don't stress over it, just me being picky lol.

Can close this one,, id rather you guys focus on other features that benefit everybody.

cheers

andynoack, 2017-04-17:
Ok, great :-) However, we will talk about that and see what we can do!

closed 2017-04-17 by andynoack