jopohl/urh #242: Fuzzing performance

andynoack, 2017-04-17:
Improve fuzzing performance for larger fuzzing labels (e.g. 16 bit)
- [ ] Add progress bar for computing fuzzed messages
- [ ] Refactor copying messages to copy on write principle, i.e. add reference to source message and save delta. If a copied+delta message is changed manually, create a real copy.
- [ ] Modulation for a large amount of messages is very time consuming. Allow to send modulated data while modulation is in progress. Prevent buffer underrun when sending.

roema, 2017-04-19:
Hi!

I have a question.
When i fuzz more than 4 bits, i get often the error 

`Traceback (most recent call last):
File "/home/roema/work/urh/src/urh/controller/GeneratorTabController.py", line 478, in on_btn_send_clicked
modulated_data = self.modulate_data()
File "/home/roema/work/urh/src/urh/controller/GeneratorTabController.py", line 352, in modulate_data
return numpy.concatenate(modulated_samples).astype(numpy.complex64)
MemoryError
`

roema, 2017-04-19:
Hi @andynoack !

I have Ubuntu 16.04 x64 with 8Gb of RAM.
I can test ist on my second desktop with 32Gb RAM.

Thanks!

roema, 2017-04-19:
Hi @andynoack !

On my second machine its works!

andynoack, 2017-04-19:
What is your setting for "Use this percentage of available RAM for buffer allocation" in Options->Device?

roema, 2017-04-19:
70%

andynoack, 2017-04-19:
Ok, then we should take a look at this!

jopohl, 2017-05-12:
fixed with [version 1.6.3](https://github.com/jopohl/urh/releases/tag/v1.6.3)

closed 2017-05-12 by jopohl

roema, 2017-05-12:
Thank you very much!!!!