jopohl/urh #244: "Using modules from /home/user/src" displayed on startup

romeojulietthotel, 2017-04-19:
Just began seeing this on urh startup. The message is displayed after:
`Will not regenerate UI, because script cant be found. This is okay in release.`

~/src has nothing to do with urh and maybe this message does not make a difference for urh.

BTW cant should be can't or cannot

jopohl, 2017-04-20:
Thanks for pointing this out. Could you test the fix I wrote?

romeojulietthotel, 2017-04-20:
Shows same path, i.e. ~/src

jopohl, 2017-04-20:
How do you run URH? From source?

romeojulietthotel, 2017-04-20:
From install:

`python setup.py install --without-hackrf --without-rtlsdr
`

then

`urh`

I haven't looked closely but the message seems to be informational only. Is that correct? Or is URH looking there for modules?

jopohl, 2017-04-21:
Well the purpose of this is to ensure, that the modules are added to the Pythonpath if URH is run from source. In your case, it can be safely ignored, if you can start URH without import errors.

romeojulietthotel, 2017-04-26:
When I first saw it I was concerned that the app was doing things in that directory. Others may draw a similar conclusion.

jopohl, 2017-04-26:
Agreed, I just updated the message to avoid confusion.

romeojulietthotel, 2017-04-26:
It's fine, looks good.

closed 2017-04-26 by jopohl