jopohl/urh #248: Open Project menu and directory tree browsing

ngmacha, 2017-04-21:
I have few comments below regarding the File menu and the directory tree browser - don't know if they are easy to fix or not, but would improve the usability of your great software! 

The File menu has New Project option, but there is no obvious way to open an existing Project... The "Open directory" offers to convert non-Project folders to Project, and opens the Project ones, however right click the folder  on the directory tree offers only "New Directory" and "Delete" - "Create/Open Project" would be very logical here!.. Also the "New Directory" should be changed to "New Folder"?..

I understand that once a Project is created, it can be reopened from the history on the bottom of File menu. However I have URH folder tree on my OneDrive, which I'm accessing from several different computers - the project created on one PC will not appear on the URH history list of another PC. An easy way to just Open Project, preferably directly from the directory tree, would help a lot. BTW, just double clicking the folder on the directory tree, even the Project folder, should not open the Project - I'm just browsing, and right-click "Open Project" when I need...

Also, once you select a folders using "Open directory" (should it be renamed to "Open Folder", for consistency with "Convert Folder to Project" below?..), there is no way to browse up the tree, even if you opt not to convert the folder to project...

Is it also possible to add a horizontal scroll to the directory tree? In many cases it's only necessary to see the right-most part of the folder/file name, and expanding the directory tree is stealing the limited screen estate.

Thank you.

jopohl, 2017-04-25:
Thanks for bringing this up @ngmacha , I will take a look at this. Just to ensure, that I do not forget anything, I will create a ToDo list here. Feel free to drop a comment, if I forgot anything or got you wrong at some point:

- [ ] add "Open Project" action to Directory Tree. BTW you can already open directories/projects from the tree by dragging and dropping them into the main application.
- [ ] rename "New Directory" action in Directory Tree to "New Folder"
- [ ] rename "Open directory" to "Open Folder" in File menu
- [ ] make tree browsable if a directory is opened
- [ ] add horizontal scroll to directory tree

ngmacha, 2017-04-27:
Thanks @jopohl , and sorry for delayed response, I'm down with a terrible flu... 
Your ToDo looks covered all. Maybe only the clarification for p.4 is to add something like "go to upper folder" will do...
 And I already see the horizontal scroll in 1.6.2.6, helps lot!

jopohl, 2017-05-12:
fixed with [version 1.6.3](https://github.com/jopohl/urh/releases/tag/v1.6.3)

closed 2017-05-12 by jopohl