jopohl/urh #249: Bit position text size (Windows)

vsboost, 2017-04-24:
Thought i would try URH in Windows just for laughs, and thought i would bring this to your attention.

jopohl, 2017-04-24:
Thanks for pointing that out, I never noticed this on Windows, seems to be related with the configured system font. Could you try setting ``` src/urh/ui/views/TableView.py:59 ``` to e.g. ``` w = self.font().pointSize() + 5 ``` and see if that works better?

vsboost, 2017-04-24:
Is that possible when using pip3 to install URH?

jopohl, 2017-04-24:
No, you would go this way:
```
git clone https://github.com/jopohl/urh/
cd urh/src/urh
./main.py
```

You can get Git for Windows [https://git-scm.com/download/win](here)

jopohl, 2017-04-24:
I just had a deeper look at it and pushed a probable fix to master. You can test it out via Git or wait till the next release, which will happen soon.

vsboost, 2017-04-24:
No probs, ill wait for a release.

vsboost, 2017-04-25:
Just to let you know, setting it to a value of 7 seemed to do the trick for me.
w = self.font().pointSize() + 7

[image]

jopohl, 2017-04-25:
Do you also need to do this with latest coder on master or is the fix I wrote sufficent?

vsboost, 2017-04-25:
Yes even with the latest code i still needed to apply the fix.

jopohl, 2017-04-25:
That's not so good, because adding an absolute offset means that the header will be too wide in other environments. Can you add a
```
print("Font Size", self.font().pointSize())
```
at  ``` urh/ui/views/TableView.py:60 ```  and post the result?

vsboost, 2017-04-25:
File "c:\users\vsboost\urh\src\urh\ui\views\TableView.py", line 60
    print("Font Size", self.font().pointSize())
                                              ^
TabError: inconsistent use of tabs and spaces in indentation

jopohl, 2017-04-25:
Please indent the new line with spaces only and not use tabs.

vsboost, 2017-04-25:
I switched themes and it looks fine in the dark theme,

Below is the standard theme
[image]

below is in the dark theme
[image]

Do you still want me to run with the print("Font Size", self.font().pointSize()) ?

jopohl, 2017-04-25:
Yes, would be very helpful. I suspect, Windows somehow bugs around and reports a pointSize of 0 to Qt. If that is the case, I could catch for 0 and fix it for the native theme.

ngmacha, 2017-04-25:
I have same problem, but I think it might have something to do with Windows10's font scaling feature, which is automatically set to 125% - 150% on the Surface tablets:
[image]

If I switch to 100% then everything displays OK...

vsboost, 2017-04-25:
Font Size 8 is what i get in the console.

Switching from (native look default) to (fallback theme) or (fallback theme dark) fixes the problem without having to exit URH however going back to (native look default) the issue reappears.

Can do a screen record if you need it.

vsboost, 2017-04-25:
I have mine set to 150% being a surface pro 2 (small screen)

jopohl, 2017-04-25:
Thanks a lot for pointing this out @ngmacha!

@vsboost : Does setting scale to 100% also solves the problem for you? At least we would know the cause then.

vsboost, 2017-04-25:
Yep, native look default works when windows scaling is set to 100%.

jopohl, 2017-04-25:
I was able to reproduce this on windows, by setting the font scaling to 150% and wrote a fix for it. Should work now.

vsboost, 2017-04-25:
Nice, working well..

Thanks again

closed 2017-04-25 by vsboost