jopohl/urh #251: You need at least Python 3.4 for this application!

muhantu, 2017-04-26:
[image]

Anyone can help me?

jopohl, 2017-04-26:
Try ``` python3.5 setup.py install ``` instead ``` python setup.py install ```.

muhantu, 2017-04-26:
ok, thanks
I'll try it..

muhantu, 2017-04-26:
[image]
this is what i got

jopohl, 2017-04-26:
You can install pyqt5 by hand if your distribution can't find it with:
```
sudo apt-get update
sudo apt-get install python3-pyqt5
```
Furthermore, you may try ``` python3 setup.py install ```, maybe ``` python3.5 ``` is not linked properly.

muhantu, 2017-04-26:
`sudo apt-get install python3-pyqt5`

that's didn't take any effect, and than I try yout suggest with python3, that's successful..
thx jopohl :D

closed 2017-04-26 by muhantu