jopohl/urh #252: Regression to Cython 0.24.1

romeojulietthotel, 2017-04-26:
I note this:

`build/lib.linux-x86_64-3.6/urh/dev/native/lib/limesdr.cpp:/* Generated by Cython 0.24.1 */`

Along with breakage.

This too:
https://github.com/jopohl/urh/commit/8eefc90378e29d47e439b948683c61b87c406b2b#diff-0d6eecfca22d708a1367b7ff3fd61f29

jopohl, 2017-04-26:
What do you mean by breakage? Is it not possible to compile limesdr.cpp anymore? Whats the error message?

romeojulietthotel, 2017-04-26:
The symptom is that the native limesdr builds but urh does not "see" the driver. i.e. no way to enable limesdr

Why the switch back to Cython 0.24.1 from 0.25.2?

jopohl, 2017-04-26:
I guess that is because @andynoack uses an older Cython version than I do. I just reverted to 0.25.2, does it work now?

romeojulietthotel, 2017-04-26:
The problem persists. I tried 1.6.2.4 also and still have the problem. I also removed and again cloned the repo but same issue.

I updated LimmeSuite so I am wondering if that broke things.

jopohl, 2017-04-26:
Possible...please do the following from the repo you cloned:
```
cd src
python3 -c "import urh.dev.native.lib.limesdr"
```
and post the result.

romeojulietthotel, 2017-04-26:
I was just trying to run test tests/TestLimeSDR.py and it could not import limesdr.

Here's output:

`urh-1.6.2.5-py3.6-linux-x86_64.egg/urh/dev/native/lib/limesdr.cpython-36m-x86_64-linux-gnu.so: undefined symbol: LMS_GetVCORange`

jopohl, 2017-04-27:
There was indeed an update in LimeSuite that causes this error. I opened #253 for and I am working on a fix, so closing this one.

closed 2017-04-27 by jopohl