jopohl/urh #254: cc1101 data de-whitening Polynom wrong

SpaceTeddy, 2017-04-29:
cc1101 de-whitening Plynom is pre set to 0x21 that means x^5+x^0.
regarding to datasheet or app note: http://www.ti.com/lit/an/swra322/swra322.pdf , Polynom should be x^9+x^5+x^0 -> 0x221

andynoack, 2017-04-29:
Well, in this case the degree of the polynomial is ignored. Does it work for you using 0x21?

SpaceTeddy, 2017-04-30:
I can not fully understand the feedback of urh.
If I send data out with 100kbit GFSK I can interprete the sniffed data till the the sync word.
i.E. snyc word 0x57 ox43; package len = 6; destination address = 03; tx_address = 01
aaaaaaaa   57435743 08083c000c30301070400040a3746d12480b7cc1  
|preamble| sync | payload

preamble and sync word is correct. Playload is strange. I disabled already FEC and payload whitening, but I can still not find the correct data. After that, I will check the whitening decoding again
chris

andynoack, 2017-04-30:
Ah ok, then I reopen the issue and we will see. Can you upload the signal you sniffed and tell where the error in the decoding is?

SpaceTeddy, 2017-04-30:
I will do so.

SpaceTeddy, 2017-04-30:
Implementation of CC1101 de-whiteing with Lfsr=0x21 is correct.
TX payload: aaaaaaaa 57435743 f9e21c9aec8730feef
decoded: aaaaaaaa 57435743 06030100010203da05

closed 2017-04-30 by SpaceTeddy

andynoack, 2017-04-30:
Great, nice to hear that :)

commit d3c28b8e (2017-04-30): Clarified format of data whitening polynomial ("w/o first bit", because it is omitted for computation of LFSR). Fix #254
commit 1c1f9042 (2017-05-05): Clarified format of data whitening polynomial ("w/o first bit", because it is omitted for computation of LFSR). Fix #254