jopohl/urh #259: LimeSDR disable

Ancer13, 2017-04-30:
[image]
  Limesdr disable and cant enable it 
Device extention still showing even restart and log out only in regular user but in sudo nothing detected

Ubuntu 16.04

jopohl, 2017-04-30:
What does 
```
locate  libLimeSuite.so
```
say?

Ancer13, 2017-04-30:
locate  libLimeSuite.so
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.02-1
/usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.02.2

jopohl, 2017-04-30:
Please do
```
sudo ln -s /usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.02.2 /usr/lib/x86_64-linux-gnu/libLimeSuite.so
```
and try again.

Ancer13, 2017-04-30:
Yes finally it works thank you very much.

closed 2017-05-01 by Ancer13