jopohl/urh #261: request: add MSK modulation type

SpaceTeddy, 2017-05-01:
i'm working with the cc1101 and this chip has different modulation types, which you can use:
ASK, 2-FSK, GFSK, 4-FSK, MSK (offset QPSK with half-sine shaping).

ASK and GFSK Mode works great, but if time please add also MSK modulation type.

thx

andynoack, 2017-05-01:
I will discuss this with @jopohl

andynoack, 2017-05-01:
Hi @SpaceTeddy 
if you are in the generator tab you can "Edit..." the modulation parameters. When choosing FSK you are able to set frequencies for 0s and 1s. If you only consider frequency values f with 1-bit=f and 0-bit=f/2, you should get a MSK modulation. Does this work for you?

SpaceTeddy, 2017-05-05:
if I change the parameters in the generator tab, does this influnce the demodulation in the interpretion section?

jopohl, 2017-05-05:
No, but in Interpretation tab you should be fine using "FSK" demodulation and using the correct center. If not, please consider providing a sample signal to us.

SpaceTeddy, 2017-05-07:
well, I tried to set the center but I had no success to get the date with FSK demudulation setting.
Therefore I've attached an urh record file. Hope you has time to check the data and give an hint to get the correct data. [https://www.dropbox.com/s/4me9dyk5dj751h6/RTL-SDR-MSK-250kBit-2M_sample.complex?dl=0](url)
---
send data: aa aa aa aa 57 43 57 43 08 03 01 00 01 02 03 04 05

thank you for your help

andynoack, 2017-05-07:
Can you increase the sample rate? It does not look like a clean record to me as the waves are cut off.

SpaceTeddy, 2017-05-07:
thanks a lot for your feedback:
attached a new record with 3M sample rate:
[RTL-SDR-MSK-250kBit-3M_sample.complex.zip](https://github.com/jopohl/urh/files/981820/RTL-SDR-MSK-250kBit-3M_sample.complex.zip)

[image]

vsboost, 2017-05-08:
The file you linked is only 12KB when extracted, should be much bigger.

jopohl, 2017-05-08:
Hi @SpaceTeddy 

did you try to demodulate the signal as PSK? Doing this for the signal you uploaded, leads to a result that seems valid to me:
[image]

andynoack, 2017-05-08:
Johannes is right, this signal looks strange somehow (might even be a PSK). Can you try to disable Frequency Correction? I think, you might be a little beside the right frequency. So tuning a little bit around 433.92M could also help to improve the output.

SpaceTeddy, 2017-05-08:
no, it does not work. I guess that MSK or offset QPSK is too far away from standard PSK or FSK.
I've to think in a different way.

[image]

andynoack, 2017-05-08:
I was not able to recover your previously sent data but I got nearer ;) It seems that it is something like FSK+Differential Encoding. But anyway, there are some bit transmission errors so you might record the signal again with another setup (change distance to the device, change frequency slightly, change frequency correction, change direct sampling?).
FYI: I used noise=0,02; center=0,1724; bitlength=9; error=5; modulation=fsk; decoding=custom (just Differential Encoding, maybe with Invert)

andynoack, 2017-05-19:
We close here, feel free to open a new issue or drop a comment if necessary.

closed 2017-05-19 by andynoack

pineappletikka, 2020-07-24:
Sorry for hijacking this issue, was wondering the same thing.
Is there any documentation around that explain a bit more in detail how to tweak URH's FSK decoder to be able to decode 2-FSK, 4-FSK, MSK, etc.?
That would be a time saver :)
Thanks

andynoack, 2020-07-26:
Actually this should work straight forward by setting Bits/Symbol (which is not autodetected) and a base modulation type like FSK, e.g. Bits/Symbol = 2 + Modulation = FSK -> 4-FSK

pineappletikka, 2020-07-29:
Playing with the Bits/Symbol value was one of the things I tried first.
https://mega.nz/folder/jURDzI5J#JkA4h95UOdBMeMHBv6K_Sw
Here I placed couple of recording I have done with my HackRF  Vs a CC1101 which was sending packets containing "HELLOWORLD" at 433MHz (I preferred to round at 433MHz in order to avoid other TX devices around messing up).
As you can see the ASK decodes fine. But 2-FSK and 4-FSK not. 
[image]

P.S. Thanks for answering this old issue anyway :)

jopohl, 2020-07-29:
Your capture suffers from quantization errors (I only checked the 3MSps/2FSK. After performing a moving average filter, the signal can be demodulated like a charm.

[image]

pineappletikka, 2020-07-30:
I am starting to see the point, thanks!
Using the following approach I was able to decode the other 2-FSK sample (at 6MSps) with some tweaks... 
e.g. 
- I doubled the samples/symbol cause the acquisition doubled from 3 to 6 MSps;
- I "smoothed" the signal with the Filter moving average button
- In Demodulated  signal view, I started moving the 1/0 Line delimiter to see if I was able to get HELLOWORLD
[image]

Though I still have a couple of missing pieces to the riddle...
1) How did you come to the conclusion of 30 Samples/Symbol? (asking cause if I try the autodect feature it tells me 100.*)
2) In the image above, (bottom part) I tried playing with filter and the other variables vs the 4-FSK signal... I got close, but not enough.

*In case of 2-FSK I may be able to infer it is indeed 30Samples/Symbol as I would do with a ASK/OOK signal: looking at the signal and measure the length of the part ( I believe being a symbol). But for 4-FSK... I am starting getting lost, visually speaking. Wonder if there's a better approach to it.

pineappletikka, 2020-08-03:
Any hint would be really appreciated. Thanks 🥰

andynoack, 2020-08-03:
You have to zoom in a little bit to see the waves. As autodetection of parameters is not supported for >1 bit/symbol you have to find the parameters manually. If you go from 2-FSK to 4-FSK the samples/symbol shouldn't have changed actually, you just see the rectangular signal on more than two different levels. Of course you have to set the 3 centers accordingly. Hope that helps!

pineappletikka, 2020-08-04:
Thanks for the hints! Will definitely take a closer look.