jopohl/urh #262: request: add FEC decoding in Decoding section

SpaceTeddy, 2017-05-01:
for transmitters, which use FEC (forward error correction) they use a Viterbi / trellis algorithm to scramble data with additional bytes.
For these kind of Payload data, it is hard to interpret the raw payload stream with urh. The cc1101 uses the following algorithm after sync word: http://www.ti.com/lit/an/swra313/swra313.pdf
If possible, please add a FEC decoding item in the Decoding menu.

example payload data:
TX payload non-encoded: aaaaaaaa 5743574 306030100010203da058
TX payload FEC-encoded: aaaaaaaa 5743574 388c83c000c3330124cf03010b8dca35340347fe38

thx,
chris

andynoack, 2017-05-01:
Hi Chris,

since FEC is quite an advanced and specialized (only CC1101?) feature, it is (at least currently) not in the basic features set. However, you are always able to call an external program from a custom decoding you create. Therefore you can write e.g. a python script or c code that decodes and encodes FEC and call it from the decoding chain automatically.

If you share your development (an external python3 script?) with us, I will consider integrating it in URH as one of the 'additional functions'. Otherwise we implement this feature when there are more people that request FEC or there is a long night without anything more fascinating to do :)

SpaceTeddy, 2017-05-01:
hi,
thanks a lot for your feedback. I've already started to do some FEC decoding analysis. If I've finished my external C tool, I can share for sure. 

BTW: congrats to urh. I like it!

SpaceTeddy, 2017-05-02:
Hi Andy,

is it possible to pass some "integer arguments" to the external program, or is is limited to one char parameter? If I try to do that, urh crashes.
example:
./test d 4 4 

thx,
Chris

andynoack, 2017-05-02:
It is not possible to pass integer parameters, as the interface is restricted to '0' and '1' chars. The format must be binary because otherwise not every sequence is possible (e.g. you cannot transmit 6 bits without trailing zeroes when using hex coding). Because of that, programs should output a binary sequence with chars.

SpaceTeddy, 2017-05-02:
here you are,
external CC1101 FEC decoding C plugin for urh.

[CC1101 FEC decoding C plugin](https://github.com/SpaceTeddy/urh-cc1101_FEC_decode_plugin)

features: 
- automatic preamble length detection
- automatic sync word length detection
- Parameters: [d | f] => "d" for output of decoded payload incl. preamble and sync word, "f" for preamble and sync word crop

unfortunately i'm not able to port this for python, sry

hoffe jemand hat damit etwas Spaß 👍 ,
Chris

closed 2017-05-02 by SpaceTeddy