jopohl/urh #264: Help for signal recording

gustavecha, 2017-05-01:
Hi,

I'm trying to record with URH a strong signal caught on SDR# but despite I'm using correct parameters, nothing seems to be recorded. 

This signal is a 12.5 kHz wide MPT1327 pulse raising every minutes.
I got it with a 30/40 dB SNR, which is good.
[image]

I can see it clearly in SDR# but nothing similar is observed in the signal recording wizard of URH (the screen below covers 1:04 minutes or recording, the MPT1327 message should have been seen withing this time range).
I don't understand what I get in URH window : all samples sound to have the same amplitude, whereas it's pretty always noise.
[image]

Obviously I'm on the same frequency than on sdr# and spectrum analyzer shows me those MPT1327 messages.
I'm using a 900 kSps sample rate with a 12.5 kHz bandwidth.

Can someone tell me what I'm doing wrong and how can I make things work ?

Thank you in advance

andynoack, 2017-05-01:
What does it look like when you use higher sample rate and bandwidth, e.g. 2M, 2M?

gustavecha, 2017-05-01:
I see little more noise, but not 30 dB pulses it seems.

Furthermore, there are several 12.5 kHz channels in 2 Mhz bandwidth, how would URH do to choose the bit to decode if so ?

andynoack, 2017-05-01:
You are right about the bandwidth, I just wanted to see whether you can see anything then. Have you tried the URH spectrum analyzer? Maybe you can use it to fine tune your parameters.

gustavecha, 2017-05-01:
Ok, understood.
I just see more noise but no signal.

On the spectrum analyzer:
I see correctly the signal I'm looking for.
I've no screen for now, but the spectrum analyzer runs like a charm, with the same parameters, on the same frequency.
With a 2 Mhz bandwidth I see all 12.5 kHz channels, as expected.

andynoack, 2017-05-02:
Difficult! I would recommend to set the bandwidth as low as possible to lower the amount of "noise" (the other channels that are transmitted in parallel). Problem here is that I do not know whether the RTL SDR is able to handle bandwidth correctly as there was an issue (missing bandwidth control) with the driver. Do you use the latest RTL SDR driver? At the moment RTLSDR via TCP is not able to set the bandwidth correctly in URH (due to driver problems, I will try to fix that...).

Furthermore you can fine tune with the noise level (by hand) for your recorded signal and hopefully you get the data you want. In this case (with several channels), our autotune for the noise level will probably not work as intented.

andynoack, 2017-05-02:
Can you test the latest code from git? I have made a little fix for rtl_tcp for bandwidth handling.

gustavecha, 2017-05-02:
Hi
Ok for the bandwidth theory, that's why I used the 900K sample rate and ~13 kHz bandwidth.

For me, RTL-SDR doesn't handle bandwidth at all but URH can filter around center freq to only get a 13 kHz window among whole 900 kHz spectrum.
It would be great to do so not only to record signal but for spectrum analyzer too where I'd like to see (like zooming in) only 20 kHz window despite I'm receiving 900k samples.
Don't you think this will save many cpu resources at runtime ?

I won't try to receive several channels at once with URH, my goal is to get a clear decoding of only one.
Multi-channel reception will be done with custom gnu-radio stuff.

I can test the new code by the end of this week, let's keep this thread active until this weekend

gustavecha, 2017-05-04:
Hi,

Tested the code on git
It doesn't change what was observed on last monday.

Tonight I see that changing gain in spectrum analyzer change something but not in signal recording window.
The level is very low, it may be the root cause making message undistinguishable from noise.
Can't it be linked to https://github.com/jopohl/urh/issues/177 ?

jopohl, 2017-05-04:
Hi @gustavecha ,

I just noticed with my RTL-SDR that the gain does not get considered if it is not set last  by URH. I just wrote a fix for this, so the latest master should work for you now.

gustavecha, 2017-05-04:
Hi Jopohl,

Tested right now and it's now fine for the level  :) :)
Thank you, it's really great

Sorry to say that again but bandwidth parameter doesn't have effect then.
When recording at a given channel central freq + 12.5kHz bandwidth, I get the same mix than 900KHz bandwidth. It mixes all 12.5 kHz channels and I'm afraid such a recording isn't valuable.
How do you feel about filtering the rtl-sdr stream since it only accepts a few sample rate values ?
May it be possible for me to post-process the recording to filter it on my own ?

andynoack, 2017-05-05:
Is there any software that allows you to record the signal as you like it to have? 
We suspect the RTLSDR driver to cause the problems here. Can you record a signal with 2M samplerate and upload it? Then we try to confirm the problem.

gustavecha, 2017-05-05:
Hi @andynoack 
I don't know any software allowing me to record what I expect to with URH, indeed.
That's why the bandwidth parameter is very interesting.

I made this chart with sdr# layout to show the difference beteen the whole rtl-sdr output and the channels i try to listen and record.
[image]

Currently, there is no intermediate filtering before recording in URH. The 900 Khz observed band is what is also seen with URH spectrum analyzer.
And then, all different channels are mixed together in the recorder and no message can finally be decoded. As you may see, many channels have the same level and will scramble themselves when no filter is used.

Since rtl-sdr won't accept bandwidth parameter out of the whole band sized by amount of sample per seconds taken, a pass-band filter have to be used to only get the 12.5 KHz bandwidth shown in the bottom left IF spectrum.

I can't use a physical filter plugged between my antenna and rtl-sdr key because the goal of my study is to setup a multi-channel decoder which listen at many of them at the same time.

I wish my knowledge of gnu-r and python signal processing was more developed to help you to make this possible. Let me know if I can provide you anything.

vsboost, 2017-05-05:
Hey@gustavecha

Have a look at Mike Ossmans hackrf lessons, in particular lesson 10 Filters
https://greatscottgadgets.com/sdr/10/

You can create a flow graph in gnuradio with the appropriate filters which will let you home in on the signal of interest. Just remember to change your source block to your rtlsdr.

andynoack, 2017-05-06:
Indeed this is a possible solution. You can implement those filters in Gnuradio (at least as long, as there are no such filters in URH...we will see) and use the the network SDR plugin (and a corresponding TCP sink in Gnuradio) to forward the pre-filtered I/Q data to URHs record signal dialog.

PR #317 (2017-08-24): Channel Feature
This PR adds a Spectrogram to Interpretation and will enable to separate frequencies bands based on this view. This makes it possible to work with signals that contain data in __multiple channels__.

This will fix the following issues:
fix #264 
fix #316 

- [x] fix bandwidth filter when selection weird ranges
- [x] spectrogram colormap selection not working (when project opened)
- [x] set loading indicator before filtering signal
- [x] GraphicViews: Select everything with CTRL+A
- [x] Use bitlength and modulation from original signal for filtered signal
- [x] small size -> cre [... cut]
closed 2017-08-24 by jopohl