jopohl/urh #268: Symbol not found: _PyString_AsString

txjacob, 2017-05-05:
I am running Sierra 10.12.4, and I have installed Python 3.6 off of the python website. However, after installing urh I get the following error when running it:

`Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/urh", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pkg_resources/__init__.py", line 33, in <module>
    import platform
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/platform.py", line 116, in <module>
    import sys, os, re, subprocess
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/subprocess.py", line 136, in <module>
    import _posixsubprocess
ImportError: dlopen(/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/_posixsubprocess.so, 2): Symbol not found: _PyString_AsString
  Referenced from: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/_posixsubprocess.so
  Expected in: flat namespace
 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/_posixsubprocess.so`

jopohl, 2017-05-05:
Please try reinstalling with
```
python3 -m pip -U --force-reinstall urh
```

txjacob, 2017-05-05:
Interestingly enough I get the same error when running that command:

`$ python3 -m pip -U --force-reinstall urh
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pip/__init__.py", line 26, in <module>
    from pip.utils import get_installed_distributions, get_prog
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pip/utils/__init__.py", line 16, in <module>
    import subprocess
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/subprocess.py", line 136, in <module>
    import _posixsubprocess
ImportError: dlopen(/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/_posixsubprocess.so, 2): Symbol not found: _PyString_AsString
  Referenced from: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/_posixsubprocess.so
  Expected in: flat namespace
 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/_posixsubprocess.so`

txjacob, 2017-05-05:
Alright, so I removed the version of python installed from the website and reinstalled python36 through macports. After I discovered the following line in my .bash_profile was causing problems with python3:
`export PYTHONPATH=/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages:/opt/local/lib/python2.7/site-packages:${PYTHONPATH}`

After removing that and reinstalling urh things are working properly. Thanks!

txjacob, 2017-05-05:
And now I can't use the BladeRF as I'm missing zeromq?

`Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urh/dev/gr/scripts/bladerf_recv.py", line 28, in <module>
    from gnuradio import zeromq
ImportError: cannot import name zeromq
`

jopohl, 2017-05-05:
Yes, seems like your GNU Radio installation has no zeromq support, you can check that with
```
python2 -c "from gnuradio import zeromq"
```

txjacob, 2017-05-05:
Yup, it appears that it got installed without it. I guess I will have to figure out how to enable it.

`python-support;testing-support;doxygen;sphinx;gnuradio-runtime;gr-blocks;gnuradio-companion;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-atsc;gr-audio;* jack;* osx;* portaudio;gr-channels;gr-noaa;gr-pager;gr-qtgui;gr-trellis;gr-uhd;gr-utils;gr-video-sdl;gr-vocoder;gr-fcd;gr-wavelet;gr-wxgui`

txjacob, 2017-05-05:
Ok, after rebuilding gnuradio with zeromq enabled, it works! Thanks so much for your help.

jopohl, 2017-05-05:
You're welcome!

closed 2017-05-05 by jopohl