jopohl/urh #270: auto-enable native backend

ZeroChaos-, 2017-05-11:
## Expected Behavior
If native backend is enabled, it should be used by default

## Actual Behavior
gnuradio backend seems to be default

## Steps to Reproduce the Problem

  1. build without  hackrf/rtlsdr/etc support
  2. start urh and check options
  3. rebuild with native supports
  4. start urh and notice options are still set to gnuradio backends

I do not see a .urh file anywhere, so maybe this is kept on the gnuradio backend by some config file, but perhaps it would be nice to track a difference between "disabled" and "the user picked this one".  like set it to "default" instead of one of the choices until the user has selected one manually perhaps?

jopohl, 2017-05-12:
The configured device backends are saved in ``` ~/.config/urh/urh.ini ```. Your assumption is correct: Currently, GNU Radio backend would be saved to ``` urh.ini ``` and prevent native backend from  being default after building native extensions.

I agree with you that this should be changed. I will have a look at it!

jopohl, 2017-05-20:
Fixed with [version 1.6.4](https://github.com/jopohl/urh/releases/tag/v1.6.4).

closed 2017-05-20 by jopohl