jopohl/urh #271: request: add GFSK modulation type

HTRLo, 2017-05-14:
Hi.
I have used URH for some projects lately and the tool makes it easy to play with unknown protocols, great work! On my latest adventure in the RF world i need to tackle a GFSK signal. Is it possible to get GFSK as a modulation in UHR? I see that it is implemented in generator part of the SW, but not in the interpretation. 
Added a cutout of my troublesome signal for reference.   
[image]

Regards Håkon

jopohl, 2017-05-15:
Could you upload a part if your signal (.complex) so we can investigate?

HTRLo, 2017-05-15:
Her is a capture of the signal.

Had to rename the zip file to PDF for GitHub to accept it. Just rename it back to .zip after download.

[RTL-SDR-155_467MHz-2MSps-2MHz.pdf](https://github.com/jopohl/urh/files/1000988/RTL-SDR-155_467MHz-2MSps-2MHz.pdf)

jopohl, 2017-05-15:
Thanks for the sample signal! We will have a look at GFSK demodulation support.

jopohl, 2017-05-18:
Hi @HTRLo ,

I played around a bit more with your signal and found out, that there is a kind of quantization error. What I did, was to put a average filter on the signal giving me this:
[image]
Does this look okay for you? I believe that URH can live without explicit GFSK demodulation support, as you can demodulate a GFSK signal by tuning the parameters in FSK demodulation as I did in the screenshot. However, I might be mistaken here! Here is the averaged signal for reference:
[averaged.pdf (rename to .complex after download)](https://github.com/jopohl/urh/files/1010659/averaged.pdf)

If this works for you, we should think about integrating an average filter into Interpretation phase.

HTRLo, 2017-05-18:
Ahh.. There is the signal ;) Tried on some other samples also, looks real promising. After some playing around i found out that 2 short pulses= binary 0 and 1 long pulse is binary 1. Now i need to try to regenerate the signal with my own bit stream to try some packet injection.

PR #277 (2017-05-28): Add DSP Filter Button to Interpretation
fix #271 
- [x] Enable Undo/Redo for Filter Operation
- [x] consider finding a better place for the button
closed 2017-05-28 by jopohl