jopohl/urh #272: RuntimeError: wrapped C/C++ object of type ZoomableGraphicView

wizz777, 2017-05-15:
#Error

ZoomableGraphicView.py", line 52, in scrollContentsBy
    super().scrollContentsBy(dx, dy)
RuntimeError: wrapped C/C++ object of type ZoomableGraphicView has been deleted
Abort trap: 6

## Steps to Reproduce the Problem

  1. Open raw file
  2. go to Generator
  3.  click Edit...
  4. Click Save and Close

## Platform Specifications

  - Python Version: 3.6
  - Operating System: MacOS 10.12.4
  - Version of URH:1.6.2.4

jopohl, 2017-05-15:
Thanks for reporting this. I wrote fix for it, so you can either check out latest master or wait for next release.

jopohl, 2017-05-20:
Fixed with [version 1.6.4](https://github.com/jopohl/urh/releases/tag/v1.6.4).

closed 2017-05-20 by jopohl