jopohl/urh #273: URH version mismatch numpy <-> compiled macOS numpy binaries?

tengelmeier, 2017-05-19:
_Please use this template for bug reports. In case of a feature request, just delete everything and write as you like._
## Expected Behavior

urh launching on mac os x

## Actual Behavior

```
>urh 
Will not regenerate UI, because script can't be found. This is okay in release.
Could not find C++ extensions, trying to build them.
Traceback (most recent call last):
  File "__init__.pxd", line 987, in numpy.import_array (src/urh/cythonext/signalFunctions.cpp:8595)
RuntimeError: module compiled against API version 0xb but this version of numpy is 0xa

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/urh/main.py", line 63, in main
    import urh.cythonext.signalFunctions
  File "src/urh/cythonext/signalFunctions.pyx", line 6, in init src.urh.cythonext.signalFunctions (src/urh/cythonext/signalFunctions.cpp:21894)
  File "__init__.pxd", line 989, in numpy.import_array (src/urh/cythonext/signalFunctions.cpp:8638)
ImportError: numpy.core.multiarray failed to import

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/site-packages/urh/main.py", line 69, in main
    os.chdir(os.path.join(src_dir, "urh", "cythonext"))
FileNotFoundError: [Errno 2] No such file or directory: '/usr/src/urh/cythonext'

```
## Steps to Reproduce the Problem

  1.  pip3 install urh
 

`pip3 install urh
Collecting urh
[....]
Installing collected packages: numpy, psutil, pyzmq, sip, pyqt5, urh
Successfully installed numpy-1.12.1 psutil-5.2.2 pyqt5-5.8.2 pyzmq-16.0.2 sip-4.19.2 urh-1.6.3`

## Platform Specifications

  - Python Version: 3.6.1  (brew installed)
  - Operating System: macOS 10.12.4
  - Version of URH: 1.6.3

jopohl, 2017-05-20:
This should be fixed with the recently released [version 1.6.4](https://github.com/jopohl/urh/releases/tag/v1.6.4). Just drop a comment here or open another issue if you experience further problems.

closed 2017-05-20 by jopohl

mosvov, 2019-01-02:
Hi guys, I have same issue in MacOS 10.14.2

> urh
Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/urh/main.py", line 84, in main
    import urh.cythonext.signal_functions
  File "__init__.pxd", line 918, in init urh.cythonext.signal_functions
ValueError: numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject

jopohl, 2019-01-02:
How did you install URH? Pip? Brew?

mosvov, 2019-01-02:
Installed via `pip3 install urh`, with latest version of python 3.
also after downgrade to v2.5.1 works without any problems

jopohl, 2019-01-02:
Can't reproduce. Here is what I did:
1. Create a fresh OSX 10.14 VM.
2. Download Python from official website.
3. ``` sudo python -m pip3 install urh ```
 a. Click through all these flashing up fancy OSX dialogs
 b. run ``` sudo python3 -m pip install urh ``` again
4. Start URH -> works

I suppose you have some varying python libraries on your system. OSX is very good at messing this up. Is it possible you installed some python libraries with brew and others with pip? This may explain the issue.

mosvov, 2019-01-02:
Yes, that can be the reason. Anyway it works now with older version, thanks for your great job!