jopohl/urh #275: uhd fails to build in 1.6.4.1 on linux

ZeroChaos-, 2017-05-21:
```
 * python3_4: running distutils-r1_run_phase distutils-r1_python_compile
/usr/bin/python3.4 setup.py --without-hackrf --without-rtlsdr --without-uhd --without-airspy --without-limesdr build
/var/tmp/portage/net-wireless/urh-1.6.4.1/temp/uhd_usrp_find9n7ab926.c:1:1: warning: return type defaults to ‘int’ [-Wimplicit-int]
 main (int argc, char **argv) {
 ^
/var/tmp/portage/net-wireless/urh-1.6.4.1/temp/uhd_usrp_find9n7ab926.c: In function ‘main’:
/var/tmp/portage/net-wireless/urh-1.6.4.1/temp/uhd_usrp_find9n7ab926.c:2:5: warning: implicit declaration of function ‘uhd_usrp_find’ [-Wimplicit-function-declaration]
     uhd_usrp_find();
     ^

Skipping native limesdr support

Skipping native airspy support

Skipping native rtlsdr support

Skipping native hackrf support

Found uhd lib. Will compile with native usrp support

Warning: Extension name 'urh.cythonext.util' does not match fully qualified name 'src.urh.cythonext.util' of 'src/urh/cythonext/util.pyx'
Warning: Extension name 'urh.cythonext.signalFunctions' does not match fully qualified name 'src.urh.cythonext.signalFunctions' of 'src/urh/cythonext/signalFunctions.pyx'
Warning: Extension name 'urh.cythonext.path_creator' does not match fully qualified name 'src.urh.cythonext.path_creator' of 'src/urh/cythonext/path_creator.pyx'
Warning: Extension name 'urh.dev.native.lib.usrp' does not match fully qualified name 'src.urh.dev.native.lib.usrp' of 'src/urh/dev/native/lib/usrp.pyx'
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --without-uhd not recognized
```

ZeroChaos-, 2017-05-21:
the actual problem appears to just be my flag, it seems --without-uhd and --with-uhd are not valid, please provide this option to override the default automagic behavior.

thanks!

jopohl, 2017-05-21:
the flags you are searching for are acutally called ``` --without-usrp ``` or ``` --with-usrp ```

ZeroChaos-, 2017-05-22:
sure is, thanks!

closed 2017-05-22 by ZeroChaos-