jopohl/urh #282: Cannot Enable HackRF Windows 7 X64

KR0SIV, 2017-06-04:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
Enable the HackRF within a Windows x64 7 environment with Python3 installed.

## Actual Behavior
The enable check mark is greyed out and I cannot enable it.

## Steps to Reproduce the Problem

  1. Install the requirements for windows
  2. run urh and goto options and devices
  3. try to enable the hackrf

## Platform Specifications

  - Python Version: python3.0.6 x64
  - Operating System:Windows 7 X64
  - Version of URH: 1.3.5

I have attempted to re-install my drivers with zadig and I have this working with the same setup on windows 10 x64.

I have tried starting the HackRF One both before and after starting the urh software.
I see now errors in my cmd promt

jopohl, 2017-06-04:
You are running a quite old version of URH. Try updating to most recent one (1.6.4.2) with:
```
pip3 install --upgrade urh
```

If it should not work after this, please post the output of:
```
python -c "import platform; print(platform.architecture()[0])"
```

KR0SIV, 2017-06-04:
Thank you, I had a feeling it was probably something wrong on my end. 
I really appreciate you taking the time to look at this for me.

This solution worked great. I had expected when I ran 'pip install urh' again it would upgrade and didn't realize I needed the flag.

Thanks again.

closed 2017-06-04 by KR0SIV