jopohl/urh #283: Cannot Start HackRF Device Windows 7 x64

KR0SIV, 2017-06-04:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
Start the HackRF successfully

## Actual Behavior
I get this error:
HackRF-SETUP: HACKRF_ERROR_NOT_FOUND (-5)

I found this odd because I have the HackRF works under SDR# and gnuradio.
I have hackrf tools installed here is the output of 'hackrf_info'

```
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0x00534f62 0x00534f62
Serial Number: 0x00000000 0x00000000 0x14d463dc 0x2f5122e1
```

## Steps to Reproduce the Problem

  1. Windows 7 x64 with requirements installed
  2. Start urh and enable the hackrf
  3. Attempt to start the device by recording a complex sample.

## Platform Specifications

  - Python Version: 3.0.6
  - Operating System: windows 7 x64
  - Version of URH: 1.6.4.2

vsboost, 2017-06-05:
Probably should update your HackRF firmware.
https://github.com/mossmann/hackrf/releases

jopohl, 2017-06-06:
Did you try updating the HackRF firmware as @vsboost suggested? You can follow [this guide](https://github.com/mossmann/hackrf/wiki/Updating-Firmware) for it. I just verified that HackRF works on Win7 with HackRF firmware version 2017.02.1.

KR0SIV, 2017-06-06:
Thank you for checking into that for me, I did update my HackRF from another pc but am unable to get to it at the moment to try again. Tomorrow I will see if it is working and close my ticket it that fixes it.

KR0SIV, 2017-06-07:
Updating the firmware did indeed solve the issue. I did not consider this would have been a problem since all was well on Windows 10 x64.

Thank you for the prompt support, I hope I wasn't too much of a bother and hope that this may help prevent others from asking the same question int he future.

KR0SIV, 2017-06-07:
Edited above comment, re-opening

vsboost, 2017-06-07:
What does device manager say for the HackRF?

KR0SIV, 2017-06-07:
HackRF One

vsboost, 2017-06-07:
Is it using Winusb driver from Zadig?

KR0SIV, 2017-06-07:
Yes, I also tried the other options to no avail

KR0SIV, 2017-06-07:
I am using the latest release of zadig

vsboost, 2017-06-07:
Can you give full output from command shell like below and try a different USB port.
-------------------------------------------
Microsoft Windows [Version 6.3.9600]
(c) 2013 Microsoft Corporation. All rights reserved.

C:\Windows\System32>urh
Using DLLs from: c:\users\xxx\appdata\local\programs\python\python36\lib\site-packages\urh\dev\native\lib\win
Will not regenerate UI, because script can't be found. This is okay in release.
Error: EnvironmentError: OSError: LoadLibrary failed to load "C:\Program Files\PothosSDR\lib\uhd\modules\umtrx.dll"

Win32; Microsoft Visual C++ version 14.0; Boost_105900; UHD_003.010.001.001-release

[INFO::SettingsProxy.py::get_receive_buffer_size] Initializing receive buffer with size 2.906GB
[INFO::Device.py::start_rx_mode] HackRF: Starting RX Mode
[←[91m←[1mERROR←[0m::Device.py::log_retcode] HackRF-SETUP: HACKRF_ERROR_NOT_FOUND (-5)
[INFO::Device.py::stop_rx_mode] HackRF: Stopping RX Mode: Stop on error
[DEBUG::Device.py::stop_tx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: Stop on error
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[DEBUG::Device.py::stop_rx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_rx_mode] HackRF: Stopping RX Mode: Dialog closed. Killing recording process.
[DEBUG::SendRecvDialogController.py::closeEvent] Device stopped successfully.
[DEBUG::SendRecvDialogController.py::closeEvent] Cleaning up device
[DEBUG::SendRecvDialogController.py::closeEvent] Successfully cleaned up device

KR0SIV, 2017-06-07:
Using DLLs from: c:\users\kr0siv\appdata\local\programs\python\python36\lib\site
-packages\urh\dev\native\lib\win
Will not regenerate UI, because script can't be found. This is okay in release.
Win32; Microsoft Visual C++ version 14.0; Boost_105900; UHD_003.010.001.001-rele
ase

[INFO::SettingsProxy.py::get_receive_buffer_size] Initializing receive buffer wi
th size 5.003GB
[INFO::Device.py::start_rx_mode] HackRF: Starting RX Mode
[←[91m←[1mERROR←[0m::Device.py::log_retcode] HackRF-SETUP: HACKRF_ERROR_NOT_FOUN
D (-5)
[INFO::Device.py::stop_rx_mode] HackRF: Stopping RX Mode: Stop on error
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[DEBUG::Device.py::stop_tx_mode] Closing parent control connection: handle is cl
osed
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: Stop on error
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread

jopohl, 2017-06-07:
Maybe there went something wrong during the update with pip? Try a fresh install:
```
pip uninstall urh
pip install urh
```

KR0SIV, 2017-06-07:
Same error after the re-installation and attempting to start my hackrf via urh

jopohl, 2017-06-07:
Ok, then we have to dig a bit deeper here. Could you please try cloning the repo and running like this:
```bash
git clone https://github.com/jopohl/urh
cd urh
cd src
python urh/cythonext/build.py
```
Now, stay in the ``` src ``` directory and tell me the outputs of the three following commands:
```python
python -c "from urh.dev.native.lib import hackrf; print('hackrf setup', hackrf.setup())"
python -c "from urh.dev.native.lib import hackrf; print('hackrf init', hackrf.init()); print('hackrf open', hackrf.open())"
python -c "from urh.dev.native.lib import hackrf; print('hackrf init', hackrf.init()); import time; time.sleep(1); print('slept 1 sec'); print('hackrf open', hackrf.open())"
```

jopohl, 2017-06-20:
I will close here, as this seems to has stalled. Drop a comment if the problem persists and I will reopen.

closed 2017-06-20 by jopohl

win10IoT, 2017-08-11:
Perhaps some additional information would help. I am encountering a similar problem (Win10 Creators) and additionally was unable to use pre-build hackrf-tools. hackrf_info returned the same above error (from libhackrf); I rebuilt libusb-1.0 and hackrf_info (not sure it matters, but I statically linked libusb-1.0) and also changed two lines in hackrf.c (only because I could not get pthread to build and hackrf.c only used 'create' and 'join' which are both one-liners in WinAPIs so replaced them with:

		DWORD tid;
		device->transfer_thread = CreateThread(NULL, 0, (LPTHREAD_START_ROUTINE)callback, device, 0, &tid);

and 

		WaitForSingleObject(device->transfer_thread, INFINITE);

(yeah, I also changed device.transfer_thread to HANDLE but again I don't think that would make any difference to what is happening)

AND yes, after rebuilding under VisualStudio 2017 Pro, my rebuilt version of hackrf_info successfully reported on the connected status of the USB device.

My first suspicion leans towards something in libusb-1.0 and a change in something maybe in the WinAPI headers... but just a first guess.

jopohl, 2017-08-11:
Thanks for the hints @win10IoT !
I just updated the dlls including libusb-1.0 in branch update_win_dlls. Could you check out that branch and see if it fixes the issue for you?

jopohl, 2017-08-11:
Works fine for me on Windows 10 Creators Update. I will merge it into master.