jopohl/urh #289: Windows platform dependencies

K80theSHADE, 2017-06-08:
To successfully install URH in Win7601 (x64) it is also necessary to install the Microsoft Windows SDK for Windows 7 and .NET Framework 3.5 SP1-->Developer Tools > Windows Headers and Libraries and Developer Tools > Visual C++ Compilers.

Without them, the installation of URH on this platform faults with the error: "    c:\program files\python35\include\pyconfig.h(243): fatal error C1083: Cannot
 open include file: 'basetsd.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\B
IN\\x86_amd64\\cl.exe' failed with exit status 2"

From this, it is clear that file, "basetsd.h" is missing.
This file is part of the Windows Headers and Libraries.
And of course if you need that you will need the 64-bit compilers.

## Platform Specifications

  - Python Version: 3.5.2
  - Operating System: Win7601 Ultimate
  - Version of URH: 1.6.4.2

jopohl, 2017-06-08:
Did you install  [Visual C++ Build Tools](http://landinghub.visualstudio.com/visual-cpp-build-tools) as stated in the Readme? I never had to install additional stuff when using that (tested on Win 7 Professional).

K80theSHADE, 2017-06-08:
I've had this for some time, but yes, I did a fresh repair.
I am doing one more repair just to be certain.

K80theSHADE, 2017-06-08:
Did not work.
Same error, even after following my own instructions in the OP....
I'm officially going to have to think about this....any additional info you need to test/TS/whatever lmk.

jopohl, 2017-06-08:
Can you explain a bit more your way of testing this? I wrote the installation instructions based on fresh Windows VMs. In your case, could there be leftovers from previous installations that produce the error you see? Here are two things that come to my mind:
- Make sure you use a combination of Python 3.5+ and Visual Studio Build Tools __2015__ as this is the recommended setup on Windows from the Python folks.
- Check your environment variables. Certain programs tend to make changes there that prevent other programs from running correctly.

K80theSHADE, 2017-06-09:
VM's, you say?
Thank you; that is a huge clue....sec.....

jopohl, 2017-06-11:
Since version 1.6.6. there are now precompiled Windows Wheels available in PyPi so you should now be able to install urh with:
``` 
pip uninstall urh
pip install urh
```
since there is no need to compile it anymore and hassle with missing header files.

jopohl, 2017-06-13:
I am closing here. Drop a comment if the problem should persist.

closed 2017-06-13 by jopohl