jopohl/urh #296: Installation

robellebor, 2017-06-21:
After installing it when i tried to open it by typing "urh" , it opens but in the command line it says like this ..

Will not regenerate UI, because script can't be found. This is okay in release.
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Wat does that mean ??

andynoack, 2017-06-21:
That means the .ui-files from QtDesigner (the tool we use to design the GUI) have already been translated to python files. You do not need to worry about this :-)

robellebor, 2017-06-21:
tnx ... and, is there any project done about gps signal using urh ????

andynoack, 2017-06-21:
I am not sure but I wanted to so the same! So, please drop us your labelsets (Project file) when you got something :)

robellebor, 2017-06-21:
will do ...

closed 2017-06-22 by jopohl