jopohl/urh #297: LimeSDR RX antenna always LNA_L

daradib, 2017-06-22:
## Expected Behavior

In spectrum analyzer or record signal, I set the antenna for LimeSDR, and the current antenna should match after I click record.

## Actual Behavior

The antenna index is set appropriately (1 for High RX_H), but the current antenna is always LNA_L.

    LimeSDR-OPEN: Success
    LimeSDR-INIT: Success
    LimeSDR-SET_CHANNEL_INDEX to 0: Success
    LimeSDR-SET_ANTENNA_INDEX to 1: Success
    LimeSDR-SET_FREQUENCY to 433920000.0: Success
    LimeSDR-SET_SAMPLE_RATE to 2000000.0: Success
    LimeSDR-SET_BANDWIDTH to 2000000.0: Success
    LimeSDR-SET_RF_GAIN to 0.5: Success
    LimeSDR: Current normalized gain is 0.50
    LimeSDR: Current antenna is LNA_L
    LimeSDR: Current chip temperature is 40.50°C
    LimeSDR: Initializing stream...
    LimeSDR-Initialize stream: Success

## Platform Specifications

  - Python Version: 3.5.3
  - Operating System: Debian 9 stretch
  - Version of URH: 1.6.6

jopohl, 2017-06-22:
Could you explain, why you are trying to use LNA_H? According to [LimeSDR documentation](https://wiki.myriadrf.org/LimeSDR_Quick_Start) LNA_L covers frequencies below 1.5 GHz which applies to your frequency of 433,92 MHz.

daradib, 2017-06-22:
I was inspecting a 2.4 GHz wireless mouse (NRF24L01 protocol?). I restarted URH so the frequency was reset to the default. I will fix it to prevent confusion.

jopohl, 2017-06-23:
How did you install LimeSuite? I have the same problem as you on ArchLinux where I compiled LimeSuite from source.
However, on a Ubuntu 16.04 VM I use the LimeSuite from [repos](http://wiki.myriadrf.org/Lime_Suite) and it works as expected. So maybe this is an option for you, to use the ubuntu repos on debian?

daradib, 2017-06-28:
liblimesuite-dev 17.02.2 works fine (after extension is rebuilt) but liblimesuite-dev 17.06.0 doesn't (after extension is rebuilt). Looks like the limesuite header files changed recently.

I'm actually using the same PPA (for xenial) on Debian rather than building from source. Does your Ubuntu VM reproduce this bug if the extension is rebuilt after installing liblimesuite-dev 17.06.0?

jopohl, 2017-06-30:
It is true, that the limesuite header files changed -- I adapted this a while ago in: 1fb008ce388aa97216826ba96ecb507bacd83859

Does the antenna selection work properly in any other software e.g. LimeSuite with version 17.06.? If so, I will have a deeper look at it, if not, we shoud consider opening an Issue in LimeSDR repo.

romeojulietthotel, 2017-08-28:
Seems like a recurring problem.  A couple of things....  @jopohl are you using the LimeSuite API in some way that makes URH fail or is the API not stable? 

I also recall that LimeSuite releases are coupled to the firmwares. But it's not clear to me how they are coupled but I do recall that if there is a mismatch between LimeSuite version and the firmware version that condition is detected and reported to stdout (maybe from SoapySDR??? or some app or libLimeSuite.so). 

Firmware directories for each version:
http://downloads.myriadrf.org/project/limesuite/

jopohl, 2017-08-28:
@romeojulietthotel you can find the LimeSDR cython wrapper [here](https://github.com/jopohl/urh/blob/master/src/urh/dev/native/lib/limesdr.pyx) and [here](https://github.com/jopohl/urh/blob/master/src/urh/dev/native/LimeSDR.py) how it is used on the python side. Basically, I just use the methods from LimeSuite in a similar way they use in their [own example](https://github.com/myriadrf/LimeSuite/blob/master/src/examples/basicRX.cpp).

So I _think_ the LimeSuite API is unstable in some way. This is supported by the fact that the URH wrapper works perfectly for some (older) versions of LimeSuite and for some not. To be more specific:
- On an Ubuntu 16.04 VM with LimeSuite installed from Ubuntu repos it works in URH like a charm.
- @andynoack is running a Gentoo PC and can also use LimeSDR in URH without problems.
- On my local Arch Linux, however, I have some problems with LimeSDR, but I can not get it to work even in LimeSuite itself. So my current workaround is to fire up my Ubuntu VM when I need to work with LimeSDR.

I would be happy for any suggestions on this.

jopohl, 2017-09-18:
This should also be fixed with 4e98ef31ffa2039f7d2ddd599dad789af7e41437.

closed 2017-09-18 by jopohl