jopohl/urh #300: New error

vsboost, 2017-07-05:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior

## Actual Behavior
Will not regenerate UI, because script can't be found. This is okay in release.
Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 112, in main
    main_window = MainController()
  File "/usr/local/lib/python3.5/dist-packages/urh/controller/MainController.py", line 52, in __init__
    project_manager=self.project_manager)
  File "/usr/local/lib/python3.5/dist-packages/urh/controller/CompareFrameController.py", line 124, in __init__
    self.field_types = FieldType.load_from_xml()
  File "/usr/local/lib/python3.5/dist-packages/urh/signalprocessing/FieldType.py", line 80, in load_from_xml
    result.append(FieldType.from_xml(tag))
  File "/usr/local/lib/python3.5/dist-packages/urh/signalprocessing/FieldType.py", line 97, in from_xml
    function = FieldType.Function[tag.get("function", "CUSTOM")]
  File "/usr/lib/python3.5/enum.py", line 277, in __getitem__
    return cls._member_map_[name]
KeyError: 'CHECKSUM'

## Steps to Reproduce the Problem

pip3 install urh

## Platform Specifications
kali linux

vsboost, 2017-07-05:
Runs fine if i run it from git download ./main.py in urh/src/urh

splotz90, 2017-07-05:
This command should help (it deletes the configured fieldtypes):

`rm -rf ~/.config/urh/fieldtypes.xml`

jopohl, 2017-07-05:
This problem is caused by running GIT master and recent release 1.6.6 in combination. The version from master writes the new ``` fieldtypes.xml ``` format which is not understood by version 1.6.6.

This will be fixed with version 1.6.7 that is aimed to be released by end of this week. Till that you may either work with master or delete ``` fieldtypes.xml ``` as @splotz90 suggested and go back to version 1.6.6. Note, however, that your custom field types will be lost if you do this. 

My recommendation is to work with the version from GIT till version 1.6.7 is released.

vsboost, 2017-07-05:
Cheers.

closed 2017-07-05 by vsboost