jopohl/urh #301: Trouble with FSK

vsboost, 2017-07-05:
[test1.cfile.gz](https://github.com/jopohl/urh/files/1126356/test1.cfile.gz)
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
In inspectrum i can see the transitions in FSK file

## Actual Behavior
In URH i can not see the transitions of the signal and am unable to make a proper decode.

## Platform Specifications

  - Python Version: 3
  - Operating System: Kali
  - Version of URH: latest from git

jopohl, 2017-07-06:
The signal has very sharp edges in the normal view and does not look like a smooth sine wave:
[image]

There are two things that you could try here:
1. Recapture the signal with a higher sample rate
2. Double check the center frequency, you may be a bit off

If that should not work let me know!

vsboost, 2017-07-06:
I will take a recapture for sure, i was just intrigued that inspectrum was able to display it.

vsboost, 2017-07-06:
[image]

vsboost, 2017-07-06:
[image]

jopohl, 2017-07-06:
That is indeed pretty impressive! You got an idea how inspectrum achieves this behind the scenes?

vsboost, 2017-07-06:
Wish i knew...

vsboost, 2017-07-06:
Think it makes use of liquiddsp

andynoack, 2017-07-07:
This issue is maybe related to #264. We will have a deeper look on it after Blackhat.

vsboost, 2017-07-07:
Cool.

jopohl, 2017-08-24:
Hey @vsboost ,
you may be interested in the new spectrogram feature of URH. I just loaded your signal and selected the desired frequency band. Then I applied a bandpass filter via context menu with a filter bandwidth of 0,001 giving me this demodulated signal:
[image]

vsboost, 2017-08-24:
Sweet.  Thanks for your hard work with this project.

closed 2017-08-24 by vsboost

michiuovidiu07, 2019-03-15:
Seara buna, am și eu o nelămurire, legat de acest subiect

michiuovidiu07, 2019-03-15:
Good evening, I also have a question about this subject

michiuovidiu07, 2019-03-15:
Please help me