jopohl/urh #320: Enabling of devices

vsboost, 2017-08-25:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
Enabled devices

## Actual Behavior
RTL-SDR / HackRF / SDRPlay are grayed out

What is the procedure to have the above devices enabled in the windows version of URH

## Platform Specifications

  - Python Version: 3
  - Operating System: Windows 8
  - Version of URH: 1.8

jopohl, 2017-08-25:
The dlls for the mentioned devices were updated for the windows version, so maybe you need to reinstall using
```
pip uninstall urh
pip install urh --no-cache-dir
```

If that should not help, you can also try reinstalling with the new ``` msi ``` installer for windows: https://github.com/jopohl/urh/releases/download/v1.8.0/Universal.Radio.Hacker-1.8.0-amd64.msi

jopohl, 2017-08-25:
Just had a deeper look on this, and I think I found the error -- might be other SDR software that interferes URH. I just fixed this in #321 and additionally added a health check button for native extenions to options. Could you pull latest master and see if it is fixed for you?

vsboost, 2017-08-25:
Cool, ill give this a go on my work computer Monday and report back.

vsboost, 2017-08-28:
Tested latest and can confirm it is working, thank you.

closed 2017-08-28 by vsboost