jopohl/urh #323: Spectrogram crashes in windows

vsboost, 2017-08-29:
## Actual Behavior
Spectrogram crashes URH

## Steps to Reproduce the Problem

Open signal
Select Spectrogram

## Platform Specifications

  - Python Version: 3.6.1
  - Operating System: Windows 8.1
  - Version of URH: 1.8.4
  - URH was installed: MSI

[image]

jopohl, 2017-08-29:
Can you upload the signal that produced this crash?

vsboost, 2017-08-29:
[RTL-SDR-433_920MHz-2MSps-2MHz.zip](https://github.com/jopohl/urh/files/1262032/RTL-SDR-433_920MHz-2MSps-2MHz.zip)

This one has no signals of interest in it but still crashes URH when spectogram is selected.

jopohl, 2017-08-30:
Thanks for the signal @vsboost that helped me to track down the error. Interestingly enough it only crashed on windows so I did not notice it when developing on linux. Should work now on latest master.

closed 2017-08-30 by vsboost