jopohl/urh #325: HackRF Windows Version

vsboost, 2017-08-30:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
HackRF Functionality 

## Actual Behavior
Can not enable HackRF

## Steps to Reproduce the Problem

Select Option
Select Devices
Select HackRF
-- Can not Enable
Click Health Check
-----------------------------------
airspy -- OK
hackrf -- ERROR: DLL load failed: The specified procedure could not be found.
limesdr -- OK
rtlsdr -- OK
usrp -- OK

INFO] Used DLLs from C:\Program Files\Universal Radio Hacker

------------------------------------

## Platform Specifications

  - Python Version: 3.6
  - Operating System: Windows 8.1
  - Version of URH: 1.8.5
  - URH was installed: via ```.msi```

jopohl, 2017-08-31:
I am unable to reproduce this on a Windows 10.

Could you post your environment variables? I have the suspicion that any of these conflicts with URHs HackRF dll.

vsboost, 2017-08-31:
USER
C:\Users\me\AppData\Local\Programs\Python\Python36\Scripts\;C:\Users\me\AppData\Local\Programs\Python\Python36\;C:\Program Files (x86)\Nmap;C:\Program Files\PothosSDR\bin

SYSTEM
C:\ProgramData\Oracle\Java\javapath;%SystemRoot%\system32;%SystemRoot%;%SystemRoot%\System32\Wbem;%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\QuickTime\QTSystem\;C:\Program Files (x86)\RTLSDR Scanner;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files (x86)\Common Files\Roxio Shared\DLLShared\;C:\Program Files (x86)\Common Files\Roxio Shared\DLLShared\;C:\Program Files (x86)\Common Files\Roxio Shared\VHStoDVD\DLLShared\;C:\Program Files (x86)\CMake\bin;C:\Program Files\SoapySDR\bin

jopohl, 2017-09-01:
These appear to the values for the ``` PATH ``` variable. They look okay, but I guess, there is a more HackRF related variable that causes trouble here. To print them just open a terminal (``` cmd.exe ```) and type ``` set ```. Here is a sample output from my windows VM
```
>set
ALLUSERSPROFILE=C:\ProgramData
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=DESKTOP-9JBENHG
ComSpec=C:\WINDOWS\system32\cmd.exe
HOMEDRIVE=C:
LOGONSERVER=\\DESKTOP-9JBENHG
NUMBER_OF_PROCESSORS=2
OS=Windows_NT
Path=C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 94 Stepping 3, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=5e03
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\WINDOWS
USERDOMAIN=DESKTOP-9JBENHG
USERDOMAIN_ROAMINGPROFILE=DESKTOP-9JBENHG
VS140COMNTOOLS=C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\Tools\
windir=C:\WINDOWS
```
could you provide me with the result of the ``` set ``` command on your machine?

vsboost, 2017-09-03:
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\me\AppData\Roaming
asl.log=Destination=file
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=mine
ComSpec=C:\Windows\system32\cmd.exe
EMC_AUTOPLAY=C:\Program Files (x86)\Common Files\Roxio Shared\
FP_NO_HOST_CHECK=NO
GRC_BLOCKS_PATH=C:\Program Files\PothosSDR\share\gnuradio\grc\blocks
GR_PREFIX=C:\Program Files\PothosSDR
HOMEPATH=\
LOCALAPPDATA=C:\Users\me\AppData\Local
NUMBER_OF_PROCESSORS=8
OS=Windows_NT
Path=C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\QuickTime\QTSystem\;C:\
Program Files (x86)\RTLSDR Scanner;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program
Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;C:\Program Files\M
icrosoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files (x86)\Common Files\Roxio Shared\DLLShared\;C:\Program Files (x86)\Common Files\Roxio Shared\DLLShared\;C:\Program Files (x86)\Common Files\Roxio\Shared\VHStoDVD\DLLShared\;C:\Program Files (x86)\CMake\bin;C:\Program Files\SoapySDR\bin; 3\;C:\Users\me\App
Data\Local\Programs\Python\Python36\Scripts\;C:\Users\me\AppData\Local\Programs\Python\Python36\;C:\Program F
iles (x86)\Nmap;C:\Program Files\PothosSDR\bin
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
POTHOS_ROOT=C:\Program Files\PothosSDR
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 60 Stepping 3, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=3c03
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Windows\system32\WindowsPowerShell\v1.0\Modules\;C:\Program Files\Microsoft Security Client\MpProvider\
PUBLIC=C:\Users\Public
SESSIONNAME=Console
SOAPY_SDR_ROOT=C:\Program Files\PothosSDR
SystemDrive=C:
SystemRoot=C:\Windows
Tech2WinUserData=C:\Users\Public\
TEMP=C:\Users\me\AppData\Local\Temp
TMP=C:\Users\me\AppData\Local\Temp
UHD_PKG_PATH=C:\Program Files\PothosSDR
USERDNSDOMAIN=DOMAIN
USERDOMAIN=DOMAIN
USERDOMAIN_ROAMINGPROFILE=DOMAIN
USERNAME=me
USERPROFILE=C:\Users\me
VOLK_PREFIX=C:\Program Files\PothosSDR
VS110COMNTOOLS=C:\Program Files (x86)\Microsoft Visual Studio 11.0\Common7\Tools\
VS120COMNTOOLS=C:\Program Files (x86)\Microsoft Visual Studio 12.0\Common7\Tools\
VS140COMNTOOLS=C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\Tools\
windir=C:\Windows

jopohl, 2017-09-04:
Looks okay to me. I even installed PothosSDR for testing and could not reproduce that behaviour. Could you run  [Dependency Walker](http://www.dependencywalker.com/) on ``` C:\Program Files\Universal Radio Hacker\hackrf.dll ``` . Should give something like this:

[image]

Maybe there is a specific dll missing on Windows 8 that comes by default in Windows 10, this way we can find out.

vsboost, 2017-09-04:
Here is what i get.

[image]

jopohl, 2017-09-05:
Interesting! LIBUSB-1.0.DLL is marked red so we may get closer. Could you expand the LIBUSB-1.0.DLL node and/or open ``` C:\Program Files\Universal Radio Hacker\libusb-1.0.dll ``` and share the results? 

Furthermore, this could indicate a corrupted installation, so reinstalling URH might be worth a try.

vsboost, 2017-09-05:
Sure.
[image]

[image]

vsboost, 2017-09-05:
Will reinstall now.

vsboost, 2017-09-05:
Found some red.
[image]

vsboost, 2017-09-05:
I deleted libusb-1.0 and hackrf dll files from URH directory and ran a repair, everything is working now.

Thanks again.

closed 2017-09-05 by vsboost

jopohl, 2017-09-05:
Cool! Glad it works now.