jopohl/urh #326: List index out of rang while testing signal decode

gustavecha, 2017-09-02:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
Use a signal to test decoding

## Actual Behavior
When selecting a signal in the list, URH stop working with error : "IndexError: list index out of range"

Traceback (most recent call last):
  File "c:\program files\python\python35\lib\site-packages\urh\controller\DecoderWidgetController.py", line 849, in set_signal
    last_message = pa.messages[-1]

## Steps to Reproduce the Problem
  1. Open Tools/Decoding
  2. Select a signal in the "test" list
  3. URH Crash

## Platform Specifications

  - Python Version: 3.5
  - Operating System: Windows 10
  - Version of URH: 1.8.5
  - URH was installed: via pip

All the best

jopohl, 2017-09-02:
Thanks for reporting this. I can confirm this is happening, when you try to change the signal in the "test" list, if the associated protocol has __no messages__. I will take care of this.

jopohl, 2017-09-02:
Fixed with 564bd73930f206c658f50d8b9abc4669b5df6657.

closed 2017-09-02 by jopohl