jopohl/urh #328: Decode CCITT #5

gustavecha, 2017-09-04:
Hi all,

i've got signals using CCITT #5 tones.
https://www.sigidwiki.com/wiki/CCITT

Is there a way to decode them with URH please ?

All the best

andynoack, 2017-09-04:
Currently URH only supports two different frequencies within a FSK. If you have a Signal with more than 1 Bit information per symbol, URH will not be able to decode it in the right way. Can you tell us about the devices you are investigating? Until now we have not had a use case for that, hence no support.

gustavecha, 2017-09-04:
CCITT #5 tones are used to select a relay in professional trunked radio networks.
Let's say it's often used as a kind of phone number in cleartext and then, messages using different modulations (at higher baud rates for instance) follow up such tones.

URH won't be able to decode all messages, obviously, but it can be really interesting to know which number is used prior to go any further

andynoack, 2017-09-05:
Sounds interesting! Do you have a small sample for us?

andynoack, 2017-12-06:
I will close here, reopen (with sample file :) if you like!

closed 2017-12-06 by andynoack