jopohl/urh #331: URH Kali Linux crash

daktar1, 2017-09-17:
Hi,
I tried to use the spectrum analyzer with hack_rf one and  kali linux. It throws an error when I try to show the signal in spectrum analyzer. After closing the window of the spectrum analyzer urh crashes. 
Did anybody success to use urh with Kali Linux?

Here is the log from the spectrum analyzer window:
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Using HackRF One with firmware 2017.02.1 
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urh/dev/gr/scripts/hackrf_recv.py", line 116, in <module>
    tb.start()
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 5654, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(r, max_noutput_items)
RuntimeError: list contains invalid format!

## Platform Specifications

  - Python Version: Python 2.7.13
  - Operating System.: Kali Linux (up to date)
  - Version of URH: 1.8.6
  - URH was installed: via pip

jopohl, 2017-09-17:
As the crash is caused by GNU Radio did you try to use the native backend for HackRF? You can change it via ``` Options -> Device ```.

If native backend is grayed for HackRF out you may need to 
```
apt-get install libhackrf-dev
``` 
and then rebuild the device extensions in the options.

daktar1, 2017-09-17:
that worked. Many thanks!

closed 2017-09-17 by daktar1