jopohl/urh #332: Clean and tighten the build environment.

romeojulietthotel, 2017-09-21:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior

Request to clean up warnings,

## Actual Behavior

Lots of warnings and other stuff that seems to be noise to my eyes.

## Steps to Reproduce the Problem

  1.  `python3 setup.py install --without-hackrf --without-rtlsdr --without-airspy --without-usrp`
  2.
  3.

## Platform Specifications

  - Python Version: python 3.6.2
  - Operating System: debian stretch
  - Version of URH: git master
  - URH was installed: __from source__ 

Examples (not complete):

`cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++`

```bundled/zeromq/tweetnacl/src/tweetnacl.c: In function ‘vn’:```
```bundled/zeromq/tweetnacl/src/tweetnacl.c:9:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]```
```#define FOR(i,n) for (i = 0;i < n;++i)```

```zmq/backend/cython/message.c: In function ‘__pyx_pf_3zmq_7backend_6cython_7message_5Frame_18set’:```
```zmq/backend/cython/message.c:3867:16: warning: variable ‘__pyx_v_buf’ set but not used [-Wunused-but-set-variable]```
```const char * __pyx_v_buf;```

> /tmp/easy_install-7j3ui_0n/pyzmq-17.0.0b1/temp/timer_createlimufwkn.c: In function ‘main’:
> /tmp/easy_install-7j3ui_0n/pyzmq-17.0.0b1/temp/timer_createlimufwkn.c:2:5: warning: implicit declaration of function ‘timer_create’ [-Wimplicit-function-declaration]
>      timer_create();
>      ^~~~~~~~~~~~
> build/temp.linux-x86_64-3.6/tmp/easy_install-7j3ui_0n/pyzmq-17.0.0b1/temp/timer_createlimufwkn.o: In function `main':
> timer_createlimufwkn.c:(.text+0x15): undefined reference to `timer_create'
> collect2: error: ld returned 1 exit status
> no timer_create, linking librt
>

jopohl, 2017-09-22:
As you mentioned elsewhere that you installed everything including python itself from source there may be some version conflicts that are normally handled by ``` pip ``` or distribution package manager. In this case, pyzmq seems to be pretty new and causes an error. This may get fixed using

```bash
apt-get install python3-pyzmq
```

or 

```bash
pip3 install pyzmq==16.0.2
```
may help here.

romeojulietthotel, 2017-09-22:
My mistake then, I knew pyzmq was part of the problem, but I thought pyzmq was included with URH.
Now I have to look and all the other warnings. Maybe they were all related to pyzmq.  I made this issue while troubleshooting as all the noise made it more difficult/time consuming wading through the output.

jopohl, 2017-09-22:
Actually I have little control on the build output at all, since I use ``` setuptools ``` for this which calls the corresponding OS compiler under the hood. Messing around with this seems a bit overkill to me.

If you share the complete log I can give you a more precise guess on what went wrong during build.

romeojulietthotel, 2017-09-26:
Thanks for the offer, this time I saved no log, and usually I save too many....

Now I am trying to figure out why the buttons have no icons.

closed 2017-09-26 by romeojulietthotel

jopohl, 2017-09-27:
Maybe you did not install an icon theme? Might be worth to try one e.g. oxygen:
```bash
apt-get install oxygen-icon-theme
```

romeojulietthotel, 2017-09-27:
That was the first thing I tried.  Seems that lightwieght desktops suffer with this problem, i.e. fluxbox, icewm, etc.  I ran in strace but haven't had time to figure it out. I noticed something funny, as an aside. URH looks everywhere for URHProject.xml. For example it looked in ~/.thumbnails. I guess it looks in every dir in ~ ???

jopohl, 2017-09-28:
Could you have a look at branch ``` iconfix ```? I added an option there to choose the bundled icons over the system icons so you should see at least the bundled ones.

jopohl, 2017-09-29:
Update: The fix is in ``` master ``` now.

romeojulietthotel, 2017-09-30:
I see icons on all the buttons now and even see more in the menus.