jopohl/urh #335: UHD Receive Stream for large sample rate/ bw causing SEGV

ianmclinden, 2017-09-28:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behavior
  * Request large-ish sample rate or bandwidth from spectrum analyzer 

## Actual Behavior
  * Python crash system alert (without crashing URH)

## Steps to Reproduce the Problem

  1. Open URH (verbose used after crash discovered)
      * python3 -vv /Library/Frameworks/Python.framework/Versions/3.6/bin/urh >> urh.py.log 2>&1
  2. File -> Spectrum Analyzer
      * Device USRP (installed and enabled under Help -> Options -> Device, status OK)
      * Freq Arbitrary, e.x. 433.920M
      * Sample Rate: 20.000M (device rating 61.44MSps)
      * Bandwidth: 20.000M (device rating 56MHz)
      * Gain: Arbitrary, e.x. 20
  3. Click Start
  4. Log window shows:
USRP-OPEN: Success
USRP: Single USRP:
  Device: B-Series Device
  Mboard 0: B200
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX1
USRP-SET_FREQUENCY to 433920000.0: Success
USRP-SET_SAMPLE_RATE to 20000000.0: Success
USRP-SET_BANDWIDTH to 20000000.0: Success
USRP-SET_RF_GAIN to 0.2: Success
USRP: Initializing stream...
USRP: Initialized stream

  5. Somewhere during the USRP-SET actions the system launches a crash report for Python
      * __pyx_pw_3src_3urh_3dev_6native_3lib_4usrp_17recv_stream causing _platform_memmove causing segmentation violation
      * [syscrash.log](https://github.com/jopohl/urh/files/1341652/syscrash.log)
  6. Freqency graph stops updating (no recv)
  7. URH Log does not show error
      * [urh.py.log](https://github.com/jopohl/urh/files/1341655/urh.py.log)

## Platform Specifications

  - Python Version: 3.6.2
  - Operating System:  Mac OS X 10.12.1 (16B2657)
  - Memory: 32GB
  - Version of URH: 1.8.7
  - URH was installed: __via pip__ 
  - SDR: Ettus B200 (over USB3.0)

jopohl, 2017-09-29:
Thanks for these detailed information! 

I tried to reproduce this behaviour with a USRP-N210 on linux, but could not provoke any error. Unfortunately, I do not possess a B200, so we cannot be sure if the error is related to this device or to OS X.

Do you have the option to try out the B200 with URH on Windows and/or Linux? This way, we may get nearer to the true error that is causing this behaviour.

ianmclinden, 2017-09-29:
I’ll try Ubuntu 16.04 /Python 3.6 today. I might have a Debian machine as well. 

I’ve only got virtualbox for Windows, and it’s usb handling is appalling, so I’m sure there will be more issues there than just the above. 

Thanks,
Ian

ianmclinden, 2017-09-29:
## Update:

Repro'd on Ubuntu, so it might be a B200 USRP issue. If you need more logs from other processes, I can grab those.

## Expected Behavior
  * Request large-ish sample rate or bandwidth from spectrum analyzer 

## Actual Behavior
  * Works the first time, crashes if stop-start toggled

## Steps to Reproduce the Problem

  1. Open URH (verbose used after crash discovered)
      * python3 -vv /Library/Frameworks/Python.framework/Versions/3.6/bin/urh >> urh_verbose.log 2>&1
  2. File -> Spectrum Analyzer
      * Device USRP (installed and enabled under Help -> Options -> Device, status OK)
      * Freq Arbitrary, e.x. 433.920M
      * Sample Rate: 20.000M (device rating 61.44MSps)
      * Bandwidth: 20.000M (device rating 56MHz)
      * Gain: Arbitrary, e.x. 20
  3. Click Start
  4. Log window shows USRP updates / log shows [O]verflow
  5. Spectrum analyzer works the first time. 
  6. Press Stop, Press Start
  7. No system crash report but spectrum analyzer stops updating
  8. URH Log does not show error
      [urh_verbose.log](https://github.com/jopohl/urh/files/1345508/urh_verbose.log)

## Platform Specifications

  - Python Version: 3.5.2
  - Operating System:  Ubuntu 16.04 LTS
  - Memory: 8GB
  - Version of URH: 1.8.7
  - URH was installed: __via pip__ 
  - SDR: Ettus B200 (over USB3.0)

jopohl, 2017-09-30:
Thanks for this test,  so now we know it is nothing OS X exclusive.

Could you try if the high sampling rates with B200 are working in other applications like GNU Radio?

ianmclinden, 2017-10-02:
Tested on the above two machines (OS X 10.12.1 & Ubuntu 16.04) 

## Test Specifications
  * Software: gnuradio-companion (3.7.10.1)
  * SDR: Ettus B200 (over USB3.0) / USRP
  * RF Center: 433.92 MHz
  * Gain: 30
  * Sample Rate: 20Msps
  * Bandwidth: 20MHz
  * Output: Waterfall GUI Sink

## Result:
  * Waterfall displays 424 MHZ - 444 MHZ correctly on both OS's
  * No [O]verflow or [U]nderflow noted in GRC log window

jopohl, 2017-10-03:
Interesting...is it also working from URH if you switch the backend for USRP to GNU Radio via Options -> Device?

jopohl, 2017-10-03:
No problem, thanks for all your information so far. So to summarize, the problem appears with native backend for USRP B200 regardless of the OS. I will see what I can do, might be an issue of buffer sizes.

Just out of curiousity: Why do you work with a high sample rate like 20M? For all devices I investigated so far, I found 2M-5M sufficent.

ianmclinden, 2017-10-03:
Summary seems correct. 

I’d also note this issue is present with any sufficiently large increase in bandwidth or sample rate, say from 1 to 2 M or from 10 to 20 M. 

I’m looking at spread spectrum devices, often using up to 75M in bandwidth. I know URH doesn’t have decode for the schemes, but the timeline tools have been useful.

ianmclinden, 2017-10-04:
## Update

I was able to recreate the bug by leaving a 1MHz / 1MSps spectrum analyzer open for a few minutes. Same issues as before, with no logged indication of an error.

jopohl, 2017-10-10:
Had a deeper look on this, but without the B200 at hand it seems impossible to locate the error. I suspect something in ``` urh/dev/native/lib/usrp.pyx:94-114 ``` causes the issue, because this is the only place where the USRP process can really segfault. Could you add some print statements to the function there, so it looks something like this:
```cython
cpdef uhd_error recv_stream(connection, int num_samples):
    print("entering receive stream")
    num_samples = (<int>(num_samples / max_num_rx_samples) + 1) * max_num_rx_samples
    cdef float* result = <float*>malloc(num_samples * 2 * sizeof(float))
    cdef int current_index = 0
    cdef int i = 0

    print("Allocating buffer")
    cdef float* buff = <float *>malloc(max_num_rx_samples * 2 * sizeof(float))
    cdef void ** buffs = <void **> &buff
    cdef size_t items_received

    print("Entering while loop")
    while current_index < 2*num_samples:
        print("Calling uhd api")
        uhd_rx_streamer_recv(rx_streamer_handle, buffs, max_num_rx_samples, &rx_metadata_handle, 3.0, False, &items_received)
        print("performing memcpy")
        memcpy(&result[current_index], &buff[0], 2 * items_received * sizeof(float))
        #for i in range(current_index, current_index+2*items_received):
        #    result[i] = buff[i-current_index]

        current_index += 2*items_received
    print("Sending result...")
    connection.send_bytes(<float[:2*num_samples]>result)
```

and then call ```  python src/urh/cythonext/build.py ``` and  run it again? This way we can at least see at which line it segfaults.

ianmclinden, 2017-10-10:
Will do, I’ll test later today.

jopohl, 2017-11-17:
Closing here, as discussion has stalled. Drop a comment if you want me to reopen.

closed 2017-11-17 by jopohl

ianmclinden, 2019-01-30:
Ref to new family of USRP B200 Issues. #608