jopohl/urh #338: merging messages, is it possible?

Nesos-ita, 2017-10-13:
i have read about the plugin to break messages but i want to do exactly the opposite.
i have recorded a few different messages of a remote temperature sensor.
what i see in analysis is:
[1] 101010101 (some kind of sync, always equal)
[2] 111011000101010101010000 (data)
[3] 101010101
[4] 101011111000110101110001 (data from other recording)

1 and 2 is a signal recorded
3 and 4 is another recored signal (when temperature changed)
the problem is that the protocol starts with 10101010... than a pause than the actual data (repeated three times with pause, not shown here) but i can't assign a label preamble/sync/whatever because it is also assigned to data.
also the checkbox "mark differences" is not very useful because we are comparing sync with data
what i'd like to have is 1 and 2 on a single line (line1) . 3 and 4 on a different single line (line 2).
in this way i can assign bits from 1to 9 as sync and the rest as data.
is it possible, am i missing something?
(urh shows "labels values for message 1")

tested so far:
-creating a new message type and applying it on line 1 and 2
-creating a partecipant and apply it to 1 and 2

partially solved in a ugly way that i'd like to avoid:
-split the transmission in many small signals (where there are pauses) for each temperature transmission so that you have a signal that is only sync, other that is only data, ...

jopohl, 2017-10-13:
Thanks for bringing this up! Just for clarification  an actual temperature message is something like:
``` sync ``` 0<sup>n</sup> ``` data ``` with n > 8
and URH misinterprets the long chain of zeros as a pause and, therefore, splits ``` sync ``` and  ``` data ``` into two separate messages?

What you could try is to shorten the long pause between ``` sync ``` and ``` data ``` so it translates to 8 or less zeros. It may be painful, however, to do this manually for a large amount of signals.

If this manual solution works for you, we could make the pause length customizable in interpretation, so that you can have messages like 101010100000000000001110 and prevent the message break.

Nesos-ita, 2017-10-15:
yes, exactly; there are about 15 consecutive zeroes.
pause length customizable is a nice idea.
Here is a photo of the whole message for completeness:
[image]

what about disabling auto split on pause or adding a checkbox to enable it?
for example what i'm doing now is recording multiple temperature+humidity transmissions and saving the whole signal.
after a bit of analysis seems that sync part doesn't change so i can edit all the saved signals and cut it away completly.
i don't know what is the supposed way/best way to use this program:
-one signal recorded=one line in the analysis (maybe with a P or a space where there is a pause instead of 0/1?)
-one signal recorded=more lines where there is a pause
this is simple transmission where noone answer so one line seems the best but if there are more devices or bidirectional transmission more lines is much better.

thanks so much for this program!
it is awesome!! <3

jopohl, 2017-10-15:
Thanks for this detailed clarification! I will make the pause length editable then, so users can control when a message break appears. If you enter 0 for this pause length this will be equivalent to disabling it completly as with the checkbox you suggested.

The intended way to use URH is to have more lines when there is a pause, so you are not forced to split signals manually. If you do not perform a line break here, it will be harder in Analysis to identify differences between messages.

Cool, that the program is helpful to you. :+1:

PR #340 (2017-10-16): fix #338 and #339
This PR fixes two issues:
- [x] fix #338 
- [x] fix #339
commit d1f2e26c (2017-10-16): fix #338 and #339

add pause threshold and store tree collapse status
closed 2017-10-16 by jopohl