jopohl/urh #341: Correct parameters for this signal

vsboost, 2017-10-16:
Hi there,

This is not a program issue but was wondering if you could possibly tell me what the correct parameters for this unknown signal is, i have tried to play around with the settings but just cant get 1's 0's from this signal, which i assume is some type of FSK signal.

[Unknown.zip](https://github.com/jopohl/urh/files/1386639/Unknown.zip)

andynoack, 2017-10-16:
There seems to be a side channel in the communication. First step would be to filter the main channel out via Signal View -> Spectrogram -> Select the main band -> Create signal from frequency selection.

Then you can smooth the new signal a bit more with the "Filter" button. Then, adjusting the parameters like in the screenshot below should be a good start.

[image]

vsboost, 2017-10-16:
Thanks Andy,

Great  tips

closed 2017-10-16 by vsboost