jopohl/urh #344: Create signal from frequency selection (Spectrogram)

vsboost, 2017-10-25:
Create signal from frequency selection is not working as expected.

[image]

[image]

[image]

jopohl, 2017-10-25:
Could you please upload a sample of  your signal so we can have a deeper look?

vsboost, 2017-10-25:
[HackRF-433_920MHz-2MSps-2MHzCopy.zip](https://github.com/jopohl/urh/files/1413871/HackRF-433_920MHz-2MSps-2MHzCopy.zip)

closed 2017-10-28 by jopohl

vsboost, 2017-10-28:
Thanks, brickwall filter works a treat.

jopohl, 2017-10-31:
Update on this: I fixed the default bandpass filter in a20ddb1d33c9c7e7d7a369cec90ff71af4587281 so it can also handle negative frequencies. Brickwall Filter may get removed soon due to bad performance.

vsboost, 2017-10-31:
No probs, look forward to testing it out.

PR #345 (2017-10-28): Add more capabilities for bandpass filtering signals
- [x] fix #344 by adding a brickwall filter
- [x] add unittest
commit 2cb7a80e (2017-10-28): add brickwall filter to fix #344