jopohl/urh #347: Cant run URH after apt upgrade

vsboost, 2017-10-28:
Did some updates on my box last night kali linux and now cant run URH

tried: pip3 uninstall urh and then pip3 install urh but still no go.

--------------------------------------------

Will not regenerate UI, because script can't be found. This is okay in release.
Could not find C++ extensions, trying to build them.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 57, in main
    import urh.cythonext.signalFunctions
ImportError: /usr/local/lib/python3.5/dist-packages/urh/cythonext/signalFunctions.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyFPE_jbuf

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 66, in main
    os.chdir(os.path.join(src_dir, "urh", "cythonext"))
FileNotFoundError: [Errno 2] No such file or directory: '/usr/src/urh/cythonext'

--------------------------------------------

However it works fine from source ./main.py

jopohl, 2017-10-28:
Can't test that right now, but according to https://stackoverflow.com/questions/36190757/numpy-undefined-symbol-pyfpe-jbuf it may help to run
``` pip3 uninstall numpy ``` and then use Kalis numpy with
```
apt-get install python3-numpy
```

or the other way around (use numpy from pip and remove numpy from Kali) if that should not work.

vsboost, 2017-10-28:
Removed numpy using pip3 but still getting same error.

running apt remove python-numpy displays this.

The following packages will be REMOVED:
  chirp dissy gnuradio gnuradio-dev gr-fosphor kali-linux-full keepnote
  killerbee python-gdal python-glade2 python-gtk2 python-matplotlib
  python-numpy python-qgis python-qgis-common python-qwt5-qt4 python-rsvg
  python-scipy python-shapely sparta xsser zenmap zim

And did not want to proceed removing all the other stuff.

tried to run pip3 install numpy and get
Requirement already satisfied: numpy in /usr/local/lib/python3.5/dist-packages

Just have to run from source for now.

jopohl, 2017-10-29:
The other stuff depends on python-numpy, but URH relies on python**3**-numpy, so ``` apt remove python3-numpy ``` might be worth a try.

vsboost, 2017-10-29:
apt remove python3-numpy

The following packages will be REMOVED:
  kali-linux-full king-phisher python3-matplotlib python3-mpltoolkits.basemap
  python3-numpy python3-pyqtgraph python3-scipy

(Reading database ... 399761 files and directories currently installed.)
Removing kali-linux-full (2017.3.2) ...
Removing king-phisher (1.7.1-0kali1) ...
Removing python3-mpltoolkits.basemap (1.0.7+dfsg-4+b2) ...
Removing python3-matplotlib (2.0.0+dfsg1-2+b1) ...
Removing python3-pyqtgraph (0.10.0-1) ...
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/tests' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/opengl/items' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/multiprocess' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/flowchart/library' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/exporters/tests' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/examples/verlet_chain' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/examples/relativity' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/examples/optics' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/dockarea' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/console' not empty so not removed
dpkg: warning: while removing python3-pyqtgraph, directory '/usr/lib/python3/dist-packages/pyqtgraph/canvas' not empty so not removed
Removing python3-scipy (0.19.1-1) ...
Removing python3-numpy (1:1.13.1-1) ...
Processing triggers for mime-support (3.60) ...
Processing triggers for desktop-file-utils (0.23-2) ...
Processing triggers for man-db (2.7.6.1-2) ...
Processing triggers for gnome-menus (3.13.3-9) ...
Processing triggers for hicolor-icon-theme (0.17-1) ...

pip3 uninstall numpy
pip3 uninstall urh

pip3 install urh
Collecting urh
Requirement already satisfied: pyzmq in /usr/lib/python3/dist-packages (from urh)
Requirement already satisfied: psutil in /usr/local/lib/python3.5/dist-packages (from urh)
Collecting numpy (from urh)
  Using cached numpy-1.13.3-cp35-cp35m-manylinux1_x86_64.whl
Installing collected packages: numpy, urh
Successfully installed numpy-1.13.3 urh-1.8.10

urh

Will not regenerate UI, because script can't be found. This is okay in release.
Could not find C++ extensions, trying to build them.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 57, in main
    import urh.cythonext.signalFunctions
ImportError: /usr/local/lib/python3.5/dist-packages/urh/cythonext/signalFunctions.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyFPE_jbuf

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 66, in main
    os.chdir(os.path.join(src_dir, "urh", "cythonext"))
FileNotFoundError: [Errno 2] No such file or directory: '/usr/src/urh/cythonext'

pip3 uninstall urh
pip3 uninstall numpy

reboot
apt install python3-numpy

urh

Will not regenerate UI, because script can't be found. This is okay in release.
Could not find C++ extensions, trying to build them.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 57, in main
    import urh.cythonext.signalFunctions
ImportError: /usr/local/lib/python3.5/dist-packages/urh/cythonext/signalFunctions.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyFPE_jbuf

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 66, in main
    os.chdir(os.path.join(src_dir, "urh", "cythonext"))
FileNotFoundError: [Errno 2] No such file or directory: '/usr/src/urh/cythonext'

No luck for me..

jopohl, 2017-10-31:
Just created a Kali VM, updated to latest version and tried to reproduce the issue. However, installation works fine for me, here is what I did:
```bash
apt-get update && apt-get -y dist-upgrade
apt-get install python3-pip
apt autoremove
apt-get install python3-psutil python3-zmq python3-pyqt5
pip3 install urh
```
A guy in [this thread](https://ubuntuforums.org/showthread.php?t=286496) mentions, that in his case this error was caused by multiple python installations, so maybe that is also the problem here? As a first start you could have a look at the output of ``` which ```. Here is what it gives on my VM:
```
root@kali:~# which python
/usr/bin/python
root@kali:~# which python3
/usr/bin/python3
root@kali:~# which pip3
/usr/bin/pip3
root@kali:~# pip3 --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)
```

vsboost, 2017-11-01:
Ran all the above and still getting.
--------------------------
root@kalisp:~# urh
Will not regenerate UI, because script can't be found. This is okay in release.
Could not find C++ extensions, trying to build them.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 57, in main
    import urh.cythonext.signalFunctions
ImportError: /usr/local/lib/python3.5/dist-packages/urh/cythonext/signalFunctions.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyFPE_jbuf

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/urh", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/urh/main.py", line 66, in main
    os.chdir(os.path.join(src_dir, "urh", "cythonext"))
FileNotFoundError: [Errno 2] No such file or directory: '/usr/src/urh/cythonext'
---------------------------------

root@kalisp:~# which python
/usr/bin/python
root@kalisp:~# which python3
/usr/bin/python3
root@kalisp:~# which pip3
/usr/bin/pip3

jopohl, 2017-11-01:
What does 
```
pip3 --version
```
say? Here is [a stackoverflow discussion](https://stackoverflow.com/questions/41878672/importing-so-file-made-with-cython-results-in-importerror-undefined-symbol) with the same question. According to this, there is something mixed up on your machine e.g. mutliple python installlations or multiple installations of the same package. 

As a last resort, you could reinstall Kali Linux, as it works without problems on a fresh system.

vsboost, 2017-11-02:
root@kalisp:~# pip3 --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)
root@kalisp:~#

jopohl, 2017-11-02:
Could there be leftovers from previous installations? Please do:
```
pip3 uninstall urh
updatedb
locate urh
```
and ensure nothing is left here.

vsboost, 2017-11-05:
Thank you, the above fixed it..

Happy days.

closed 2017-11-05 by vsboost