jopohl/urh #348: Substitution

vsboost, 2017-10-28:
Hi there,

Here is what i have: 100000100000100010001000100000100010001000100000
Data is either 100000 or 1000

I was trying to use substitution to convert 100000 to 1 and 1000 to 0 but this does not work, unless i'm not using it correctly.
[image]

Is there any other way you recommend having this data converted as it would make fuzzing much easier if you only deal with 1's or 0's that are really 100000 or 1000

Hope this makes sense.

andynoack, 2017-10-28:
Hey @vsboost,
I think, you can solve that by:
1.) Invert
2.) Morse Code (3,4,1)
Does that help you?

vsboost, 2017-10-28:
Yes this works.. Thanks for that.

closed 2017-10-28 by vsboost