jopohl/urh #349: Feature request: Add usb oscilloscope and/or logic analyzers as signal source.

shumtravi, 2017-10-31:
Please, can you add these feature to make possible using of popular usb oscilloscopes (DScope, Hantek etc) as signal sources for low freq (down to DC) or just demodulated signal protocol analysis. For example we can use it to extract signal from near electromagnetic field _(coil/optic sensor)->oscilloscope->urh_. Or may be just ability to import saved data from scopes?

jopohl, 2017-11-01:
Sounds indeed pretty cool! However, I did not work with such a usb oscilloscope and we do not possess any to perform tests. So I think it would be more easy to add the ability to import saved data from scopes. 

Can you tell me a bit more about the file format such scopes use?

shumtravi, 2017-11-01:
I have only DScope oscilloscope from [Dreamsourcelab](http://www.dreamsourcelab.com/) it uses open source software [DSView](https://github.com/DreamSourceLab/DSView) and can export data to .csv format.

Just one example. I have arduino running simple code that connects with another device with 1-wire interface. I made two coils connected to transformer as differential electromagnetic probe for DSO to investigate ability of sniffing that conversation by collecting near field noise. It can be just one coil without transformer, but with two coils we can add signal or subtract some noise from them by adjusting them in space near device. Or we can use expensive current probes for oscilloscopes.

It's my captured data in CSV (just rename it from pdf):
[1-wire-full.pdf](https://github.com/jopohl/urh/files/1435573/1-wire-full.pdf)

Sensor setup:
[image]

Here is screenshot of full message. Yellow channel is collected noise and green is real signal from 1-wire bus. As we can see 0 and 1 levels distinguishable by background noise level and edges looks like differential spikes on noise diagram.
[image]

On zoomed view we can detect rising/falling edge of original signal as sign of first half-wave of damped oscillation generated in the oscillatory circuit (coil with parasitic capacitance of cable).
[image]

I have not any other oscilloscope right now but i think they can export data to csv too.

closed 2017-11-02 by jopohl

shumtravi, 2017-11-13:
Thank you, it works, but i get .csv data from another osciloscopes (Tektronix and Agilient) and it have slighty another format. Dslogic presents data with sample rate in header and data array in volts after that (one column for channel), but Tektonix and Agilient represents signal with two columns (timestamp and discrete value). Here is two captures renamed to pdf:

[Agilient-DSO-X-2002A.pdf](https://github.com/jopohl/urh/files/1465784/Agilient-DSO-X-2002A.pdf)
[Tektronix-MSO-2024.pdf](https://github.com/jopohl/urh/files/1465785/Tektronix-MSO-2024.pdf)

jopohl, 2017-11-13:
I see...so there is only one channel for Tektonix and Agilent? So if we map it to a IQ signal, Q would be zero for all samples?

shumtravi, 2017-11-13:
Yes you are right, there is only one channel presented for Tektronix and Agilent, btw in first capture by DScope is I-signal and zero-Q too, but two channels presented. 1I and 1Q=0 for 1 ch. with our signal and 2I 2Q=0 for second channel, but 2-ch needs to be dropped in DScope capture, i present it only as "original" signal that needs to be reconstructed from ch1 signal. Just for explanation. We need drop ch-2 in first capture.

jopohl, 2017-11-13:
So, in fact, __all__  oscilloscopes only use one channel? If that is the case I can modify the csv import. 

Tektronix and Agilent both use a timestamp in their csv. Can you also get Dscope to produce something like:

| Timestamp | Data |
| ------------- | ------------- |
| Timestamp 1  | data 1  |
| Timestamp 2 | data 2  |

In this case, we can always take the second column as data.

shumtravi, 2017-11-13:
Actually yes, all oscilloscopes use one "I" channel to present data from physical channel. I can't change format that DScope uses to save data and that two types of captured data presentation both popular on different scopes . May be best way is create something like universal "import csv data" wizard, that shows header of the file and first several strings of data, to ask user which column to use as data, which column for timestamp (and calculate sample rate), or let em input sample rate itself? I think that kind of import wizard will be universal for both types of files and any column can be used as data input (for example from multi channel capture, without any pre-editing .csv file).

jopohl, 2017-11-13:
Sure thing, if both formats can be found in the wild an import wizard would be a nice way to deal with this. I also like the idea of calculating the sample rate from timestamps. I will open a new issue for these enhancements and implement them soon.

shumtravi, 2017-11-13:
Thanks a lot

PR #353 (2017-11-02): Add import for csv
fix #349
commit 70d8968a (2017-11-02): Add import for csv (#353) to fix #349

* add import for csv

* make codecov run instantly

* do not wait for appveyor with codecov

* add unittest for csv import

* clean up after yourself

* use codecs to open csv file

* use csv module for csv reading

* rename variable

* update coveragerc
commit bd6b8e0a (2017-12-04): Add import for csv (#353) to fix #349

* add import for csv

* make codecov run instantly

* do not wait for appveyor with codecov

* add unittest for csv import

* clean up after yourself

* use codecs to open csv file

* use csv module for csv reading

* rename variable

* update coveragerc