jopohl/urh #352: Add possibility to set a minimum lenght for a message

inode-, 2017-11-02:
During my tests I saw that the software is not able to manage the 0 at the end of the message in case of ASK demodulation. For example if our message is 100100101010101100 the software will interprete this as: 1001001010101011.

I didn't find any way to force the software to set a minimum lenght of a message that should be interpreted.

jopohl, 2017-11-02:
Could you upload a signal that has this behaviour?

vsboost, 2017-11-05:
I have come across this as well, would be nice to set a minimal message length so if the signal ends with a 1 to add the necessary amount of 0's to complete the message.

inode-, 2017-11-05:
You can download a sample here: https://mega.nz/#!ncN0mb4Q!h4aGhHtdqHxELCvYCDgM9lSlV8--VxAzRqx0HY-rxN0 the modulation is OOK (an ASK with trasmission or no trasmission) with PWM encoding, 100 = 1, 110 = 0

jopohl, 2017-11-07:
I hear you guys,  that would greatly simplify the demodulation of such signals. Will take care of it soon.

PR #354 (2017-11-13): make minimum message length configurable
Goal of this PR is to fix #352 .

- [x] add a new dialog for advanced modulation options
- [x] implement logic for message divisor length
- [x] ensure ``` pause_threshold ``` and ``` message_divisor_length ``` are considered in ``` apply to all ``` action
- [x] ensure both parameters are saved in project file
- [x] make advanced settings button only visible for ASK demod
- [x] unittests
   - [x] pause threshold
   - [x] message length divisor
closed 2017-11-13 by jopohl