jopohl/urh #355: Enhance csv import

jopohl, 2017-11-13:
Based on the discussion in #349 enhance the csv import in the following ways:
- [ ] show an import wizard with preview of first columns and let users choose which column to use as  I-data 
- [ ] Q data is always zero for oscilloscopes, so choosing Q data should be optional in the wizard
- [ ] If timestamp column is given and selected, calculate sample rate from timestamps

closed 2017-11-15 by jopohl

shumtravi, 2017-11-17:
Please add field "Sample rate". Now importer works fine with files with timestamps, but not so clear how it works when sample rate  already presented in header without timestamp column (With DScope-like files). May be you can made that field automatically calculated when timestamp column presented and manually editable if it not presented?

jopohl, 2017-11-18:
I wrote that in the tooltip for the timestamp column: You can manually edit the sample rate in the signal details after import. The signal details can be opened with this button:

[image]

Does this work for you, or is it crucial to have a sample rate fiel in the CSV dialog?

shumtravi, 2017-11-18:
Oh, now i see, it works, thank you. I think with sample rate field it will looks just more clear, but it not so crucial for me. Anyway your new csv-wizard looks good.

PR #357 (2017-11-15): Enhance CSV Import
fix #355 

- [x] make filename configurable in dialog
- [x] if csv file is opened from regular dialog -> open import dialog with preselected filename
- [x] add error handling if filename not found
- [x] open dialog on Import Action
- [x]  If timestamp column is given and selected, calculate sample rate from timestamps
- [x] add unittests for dialog