jopohl/urh #359: Error opening sample: Unpack requires a bytes object of length

GreenSlime, 2017-11-16:
_Please use this template for bug reports. If you have a feature request or question just delete everything and write as you like._
## Expected Behaviour
open .wav file

## Actual Behaviour
Errors out, image of dialogue added to post.

## Steps to Reproduce the Problem

  1. open URH
  2. Open file *.wav
  3.

## Platform Specifications

  - Python Version: 3.6
  - Operating System: Win 10 x64
  - Version of URH: 1.8.12
  - URH was installed:  MSI x64
[image]

Most likely an issue with my environment, a few clues on what to investigate would be useful.

jopohl, 2017-11-16:
Only single channel wavs are supported, which may cause this error. This is documented [in the wiki](https://github.com/jopohl/urh/wiki/Supported-signal-file-formats).

Is it possible for you do recapture the signal and save it as ``` .complex ``` from URH or from the application you created the ``` .wav ``` ?

GreenSlime, 2017-11-17:
File was produced with SDR#, stripping a channel off the sample still produces the same error, but I imagine I need to change my approach rather than improvise. 
I will use URH to gather the sample instead. A pity, as the signal I am monitoring is very sporadic, and SDR# made it convenient to monitor the frequency and gather a sample.
It looks like I would be better with GQRX perhaps.

Thank you for your advice

jopohl, 2017-11-17:
Could you provide me with sample of your .wav file? I might add an import for SDR# created wavs.

jopohl, 2017-11-18:
I just updated the file open routine for WAV files so it should be no problem anymore to open a WAV file from SDR#.

PR #362 (2017-11-18): Improve handling of WAV files
fix #359

- [x] Add support for two channel wavs
- [x] adapt documentation in wiki
- [x] drop question "already quad demod?" when opening a wav
closed 2017-11-18 by jopohl